m-labs/artiq #941: Startup kernels with DRTIO

cjbe, 2018-03-02:
Startup kernels are currently problematic with DRTIO. The startup kernel runs before the DRTIO link(s) are up. If the startup kernel uses remote RTIO it locks up until the link(s) are up, then gives an underflow error. If the link(s) do not come up, no error is emitted and the master hangs.

Startup kernels that use remote RTIO are very useful for initialising hardware (e.g. Urukul, TTL states, etc.), so it would be useful to work out some robust way of doing this.

cjbe, 2018-03-02:
I appreciate the difficulty here.

How about:
1) For the time being, DRTIO operational means all links defined at compile time are up
2) In the future: A coreconfig entry defines the desired tree.

jordens, 2018-03-02:
Or maybe expose that info to kernels.

sbourdeauducq, 2018-03-03:
Yes, maybe the startup kernel can explicitly wait for all relevant links to be up. I propose introducing a separate API call to check for link status, which will be cleaner than attempting RTIO operations until ``RTIOLinkError`` (#942) is no longer raised.

commit 432e61bb (2018-03-04): drtio: add kernel API to check for link status. Closes #941
closed 2018-03-04 by sbourdeauducq