m-labs/artiq #943: try/finally on kernel corrupts passed-through host exception name

klickverbot, 2018-03-03:
```python
from artiq.experiment import *

class TestFinally(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    def run(self):
        self.kernel_fun()

    @kernel
    def kernel_fun(self):
        try:
            self.host_fun()
        finally:
            print("finally")

    def host_fun(self) -> TNone:
        raise ValueError
```

gives (on 3505878)

```
finally
Traceback (most recent call last):
  File "/home/ion/miniconda3/envs/artiq/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/ion/scratch/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/ion/scratch/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/ion/scratch/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "test_finally.py", line 8, in run
    self.kernel_fun()
  File "/home/ion/scratch/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/ion/scratch/artiq/artiq/coredevice/core.py", line 133, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/home/ion/scratch/artiq/artiq/coredevice/comm_kernel.py", line 606, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/home/ion/scratch/artiq/artiq/coredevice/comm_kernel.py", line 589, in _serve_exception
    core_exn = exceptions.CoreException(name, message, params, traceback)
  File "/home/ion/scratch/artiq/artiq/coredevice/exceptions.py", line 21, in __init__
    self.id = int(exn_id)
ValueError: invalid literal for int() with base 10: 'n'
```

This is because the exception `name` is `n:ValueError` instead of the exception id (i.e. `0:ValueError`). I haven't investigated further – memory corruption?

whitequark, 2018-03-03:
Probably [this](https://github.com/m-labs/artiq/blob/master/artiq/firmware/ksupport/eh.rs#L403-L405).

whitequark, 2018-03-03:
This is exceptionally (no pun intended) annoying to fix. We already take a copy of the exception itself but it's not easily possible to take a copy of data without an allocation of some sort. Maybe put the data into runtime (not kernel) memory...

klickverbot, 2018-03-03:
(cc @kesht123)

sbourdeauducq, 2022-01-25:
The particular corruption issue is fixed, but it seems the ``finally`` block is not run, or the ``print`` RPC doesn't occur for some other reason. The full output is:
```
Core Device Traceback:
Traceback (most recent call first):
  File "foo.py", line 18, in host_fun
    raise ValueError
  File "foo.py", line 13, in ?? (RA=+0x188)
    self.host_fun()
ValueError(0): 

End of Core Device Traceback

Traceback (most recent call last):
  File "/nix/store/j6by8h97dv43br92qf01gxj0v6q4ih3f-python3.9-artiq-7.7940-9d43762-beta/bin/.artiq_run-wrapped", line 9, in <module>
    sys.exit(main())
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/frontend/artiq_run.py", line 224, in main
    return run(with_file=True)
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/frontend/artiq_run.py", line 210, in run
    raise exn
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/frontend/artiq_run.py", line 203, in run
    exp_inst.run()
  File "foo.py", line 8, in run
    self.kernel_fun()
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/language/core.py", line 54, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/coredevice/core.py", line 139, in run
    self.comm.serve(embedding_map, symbolizer, demangler)
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/coredevice/comm_kernel.py", line 698, in serve
    self._serve_exception(embedding_map, symbolizer, demangler)
  File "/nix/store/dajdslg1skr16258kgya7c0mnx1y6avm-python3-3.9.6-env/lib/python3.9/site-packages/artiq/coredevice/comm_kernel.py", line 680, in _serve_exception
    raise python_exn
ValueError
```

pca006132, 2022-01-27:
> The particular corruption issue is fixed, but it seems the `finally` block is not run, or the `print` RPC doesn't occur for some other reason. The full output is:
>
> ```
> Core Device Traceback: [...]

As we cannot reproduce this, should we close this issue?

closed 2022-01-27 by sbourdeauducq