m-labs/artiq #944: Sayma_AMC with SAWG congestion Level 5

jbqubit, 2018-03-04:
I don't know when this started but Vivado reports [congestion level 5](https://www.xilinx.com/support/answers/66314.html). This looks to increase build time and Vivado says it can "impact timing closure." 

```
INFO: [Route 35-449] Initial Estimated Congestion
 ________________________________________________________________________
|           | Global Congestion | Long Congestion   | Short Congestion  |
|           |___________________|___________________|___________________|
| Direction | Size   | % Tiles  | Size   | % Tiles  | Size   | % Tiles  |
|___________|________|__________|________|__________|________|__________|
|      NORTH|     4x4|      0.19|     2x2|      0.27|     4x4|      0.85|
|___________|________|__________|________|__________|________|__________|
|      SOUTH|   32x32|      5.49|   32x32|      6.47|   32x32|      6.82|
|___________|________|__________|________|__________|________|__________|
|       EAST|     4x4|      0.29|     4x4|      0.52|     8x8|      1.72|
|___________|________|__________|________|__________|________|__________|
|       WEST|     2x2|      0.21|     2x2|      0.35|     8x8|      1.46|
|___________|________|__________|________|__________|________|__________|
Congestion Report
GLOBAL Congestion bounded by tiles (Lower Left Tile -> Upper Right Tile):
SOUTH
	INT_X32Y188->INT_X47Y219 (XIPHY_L_X32Y180->CLEL_R_X47Y219)
	INT_X32Y204->INT_X47Y219 (XIPHY_L_X32Y180->CLEL_R_X47Y219)
	INT_X32Y188->INT_X47Y203 (XIPHY_L_X32Y180->CLEL_R_X47Y203)
LONG Congestion bounded by tiles (Lower Left Tile -> Upper Right Tile):
SOUTH
	INT_X32Y164->INT_X55Y251 (XIPHY_L_X32Y120->CLEL_R_X55Y251)
	INT_X32Y220->INT_X47Y235 (XIPHY_L_X32Y180->CLEL_R_X47Y235)
	INT_X32Y204->INT_X47Y219 (XIPHY_L_X32Y180->CLEL_R_X47Y219)
	INT_X32Y188->INT_X47Y203 (XIPHY_L_X32Y180->CLEL_R_X47Y203)
SHORT Congestion bounded by tiles (Lower Left Tile -> Upper Right Tile):
SOUTH
	INT_X40Y156->INT_X47Y219 (CLE_M_X40Y156->CLEL_R_X47Y219)
	INT_X32Y204->INT_X47Y219 (XIPHY_L_X32Y180->CLEL_R_X47Y219)

INFO: [Route 35-448] Estimated routing congestion is level 5 (32x32). Congestion levels of 5 and greater can reduce routability and impact timing closure.
```

sbourdeauducq, 2018-03-04:
Yes, we know about this. Did it not meet timing?

sbourdeauducq, 2018-03-10:
#951

closed 2018-03-10 by sbourdeauducq

jbqubit, 2018-03-12:
Vivado is working very hard to achieve closure and seems on the edge of not finding a solution. If the Xilinx model for Kintex UltraScale were slightly off, Vivado may claim timing closure but produce a .bit that is borderline. Do you have reason to suspect this? 

I'm uncomfortable with Sayma's reliance on a gateware design that's so brittle. What are the prospects for reducing congestion in Sayma with SAWG? What steps have you taken to improve the situation? @hartytp or others have suggestions for other things to check. Is a larger FPGA needed?

sbourdeauducq, 2018-03-12:
Kintex-7 doesn't seem to have this problem; at least we did not observe it (though the SAWG on KC705 has fewer channels).

hartytp, 2018-03-12:
> If the Xilinx model for Kintex UltraScale were slightly off, Vivado may claim timing closure but produce a .bit that is borderline. Do you have reason to suspect this?

Shouldn't do if everything is correctly constrained. Vivado is supposed to perform a conservative timing analysis that provides a complete guarantee over PVT. There is always the possibility of a Vivado bug, but we have used a few different versions.

>  Is a larger FPGA needed?

I still suspect we're missing something obvious and that there is a relatively simple gateware solution.

Bigger FPGA probably not the way to go (the compile time is already long enough). Simplifying the SAWG in the way @jordens suggested elsewhere could help quite a bit and is probably worth doing in general.