m-labs/artiq #947: Intermittant DRTIO write underflows

cjbe, 2018-03-05:
I occasionally (a few times a day) see a stream of 'write underflow' errors from the Kasli satellite (using a branch of Artiq a274af77).

I seem to get them most frequently with an (underflowing) DMA sequence like the below experiment, but I have seen them without using DMA. The probability of seeing this error seems to be changed by restarting the master+satellite - i.e. sometimes I don't see this error for a long time, then after restarting everything I see it straight away.

```python
self.core.reset()
with self.core_dma.record("x"):
    for _ in range(1000):
        self.rttlo0.pulse_mu(8)
        delay_mu(8)
h = self.core_dma.get_handle("x")
self.core.break_realtime()
self.core_dma.playback_handle(h)
```

Satellite log:
```
[     0.000006s]  INFO(satman): ARTIQ satellite manager starting...
[     0.004622s]  INFO(satman): software version 4.0.dev+670.ge3938620
[     0.010802s]  INFO(satman): gateware version 4.0.dev+670.ge3938620
[     1.977050s]  INFO(board_artiq::si5324): Si5324 is locked
[     2.125206s]  INFO(satman): link is up, switching to recovered clock
[     3.794707s]  INFO(board_artiq::si5324): Si5324 is locked
[     3.798796s]  WARN(satman): aux packet error (gateware reported error)
[    14.425707s] ERROR(satman): write underflow
[    14.428562s] ERROR(satman): write underflow
[    14.432730s] ERROR(satman): write underflow
[    14.436898s] ERROR(satman): write underflow
[    14.441066s] ERROR(satman): write underflow
[    14.445234s] ERROR(satman): write underflow
...
```

sbourdeauducq, 2018-03-05:
How are you clocking the master and satellite?

cjbe, 2018-03-05:
The master is running from an external 150 MHz. The slave is starting off on the external 150 MHz and switching to the recovered clock when the DRTIO link is up.

sbourdeauducq, 2018-03-05:
Data corruption from the Si5324 non-deterministic skew? Let's revisit this issue after the latter is fixed.

sbourdeauducq, 2018-03-08:
Reproduced.

sbourdeauducq, 2018-03-08:
Ran the repro several hundred times after the fix without any error.
What kind of fiber lengths are you going to use DRTIO with? If they vary a lot we should do a RTT measurement and compute the underflow limit value in the firmware.

sbourdeauducq, 2018-03-08:
(The value we had before was for GTX+62.5MHz clock with a short fiber, which is why we are seeing this problem now with the 150MHz clock)

klickverbot, 2018-03-08:
Most of our fibres will be 10m or 15m, but there might be the occasional in-rack or ~30 m link in the future.

cjbe, 2018-03-08:
My preference would be for an RTT measurement. As @klickverbot says, links will typically be 2m - 15m, but up to (at least) 30m on occasion.

sbourdeauducq, 2018-03-09:
Thinking more about this, it's not that simple:
* The timestamp sent during TSC synchronization goes the same way as the writes, so it's already compensated.
* Correcting the timestamp automatically based on RTT measurements (so that both the satellite and the master have the same TSC, not a delayed version of it) is subject to +/- 1 cycle non-determinism.
* Timestamp correction at the DRTIO level is redundant with #40.
* #40 would also provide the higher resolution of fine RTIO cycles, whereas DRTIO timestamp correction could only use coarse RTIO cycles.

I propose moving all of this into #40, i.e. the latency adjustments of #40 should also change the write underflow margins of their respective DRTIO links.

cjbe, 2018-03-11:
With fc3d97f I still see infrequent write underflows. With the following experiment, I see a write underflow every 5 minutes or so.

```python
class RepeatingPulses(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttlo0")
        self.setattr_device("ttlo1")
        self.setattr_device("rttlo0")

    @kernel
    def run(self):
        self.core.reset()

        while True:
            delay_mu(160000)
            self.ttlo1.pulse_mu(8)
            for _ in range(4):
                delay_mu(8)
                with parallel:
                    with sequential:
                        delay_mu(200)
                        self.ttlo0.pulse_mu(80)
                    self.rttlo0.pulse_mu(80)
```

```
...
[182672.828990s] ERROR(satman): write underflow
[182728.965752s] ERROR(satman): write underflow
[182744.736193s] ERROR(satman): write underflow
[183313.054034s] ERROR(satman): write underflow
[183633.298499s] ERROR(satman): write underflow
[183668.369553s] ERROR(satman): write underflow
[184508.823617s] ERROR(satman): write underflow
[184793.470987s] ERROR(satman): write underflow
[185050.086677s] ERROR(satman): write underflow
[185330.952187s] ERROR(satman): write underflow
[185371.287073s] ERROR(satman): write underflow
[185416.710975s] ERROR(satman): write underflow
[185453.938582s] ERROR(satman): write underflow
[186011.241196s] ERROR(satman): write underflow
[186053.002078s] ERROR(satman): write underflow
```

sbourdeauducq, 2018-03-12:
@cjbe Are the two local TTLs necessary to reproduce the bug?

cjbe, 2018-03-12:
Removing the local TTLs and making the remote output events more frequent increases the rate of errors to ~ 1/s on my setup, which consists of 2x Kasli v1.0 connected by a 10m fiber:
```
...
[    51.320263s] ERROR(satman): write underflow
[    51.440848s] ERROR(satman): write underflow
[    51.609055s] ERROR(satman): write underflow
[    52.146008s] ERROR(satman): write underflow
[    53.479435s] ERROR(satman): write underflow
[    53.733276s] ERROR(satman): write underflow
[    55.101214s] ERROR(satman): write underflow
[    57.201845s] ERROR(satman): write underflow
```

```python
class RepeatingPulses(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttlo0")
        self.setattr_device("ttlo1")
        self.setattr_device("rttlo0")

    @kernel
    def run(self):
        self.core.reset()

        while True:
            delay_mu(16000)
            for _ in range(4):
                delay_mu(8)
                with parallel:
                    with sequential:
                        delay_mu(200)
                    self.rttlo0.pulse_mu(80)
```

sbourdeauducq, 2018-03-12:
I can reproduce this bug; interestingly the underflows have positive slack:
```
[   303.639475s] ERROR(satman): write underflow, channel=1, timestamp=317310308344, counter=317310042128, slack=266216
[   304.955047s] ERROR(satman): write underflow, channel=1, timestamp=318888999000, counter=318888738832, slack=260168
[   308.987667s] ERROR(satman): write underflow, channel=1, timestamp=323728181832, counter=323727916944, slack=264888
[   311.518213s] ERROR(satman): write underflow, channel=1, timestamp=326764859496, counter=326764593040, slack=266456
[   311.699091s] ERROR(satman): write underflow, channel=1, timestamp=326981908760, counter=326981648400, slack=260360
[   312.450565s] ERROR(satman): write underflow, channel=1, timestamp=327883683240, counter=327883423632, slack=259608
[   312.680377s] ERROR(satman): write underflow, channel=1, timestamp=328159458728, counter=328159199120, slack=259608
[   312.823681s] ERROR(satman): write underflow, channel=1, timestamp=328331430776, counter=328331165584, slack=265192
```

sbourdeauducq, 2018-03-12:
Been running for a few minutes without any error - before that commit there was an error every few seconds.

closed 2018-03-12 by sbourdeauducq

cjbe, 2018-03-13:
This works for me - I see no errors over 12 hours.

commit e38187c7 (2018-03-08): drtio: increase default underflow margin. Closes #947