m-labs/artiq #950: sequence errors with idle kernels

jordens, 2018-03-09:
* Single channel, led0 only
* Continuously increasing timestamps
* Idle kernel and experiment are the same
* When idle kernel is interrupted and experiment kernel start *or* when experiment kernel is ctrl-c-ed and when idle kernel starts, there is a sequence error on channel 32 (led0).
* Current master, opticlock, v1.1, Si5324 clock.

```python
from artiq.experiment import *

class Idle(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led0")

    @kernel
    def run(self):
        self.core.break_realtime()
        while True:
            self.led0.pulse(.5*s)
            delay(.5*s)
```

```
artiq_compile -o repository/test_idle.elf repository/test_idle.py
artiq_coreconfig write -f idle_kernel repository/test_idle.elf
# wait a bit
artiq_run repository/test_idle.py
```

```
[   388.759626s]  INFO(runtime::session): no connection, starting idle kernel
[   388.791076s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 32
[   389.821390s]  INFO(runtime::mgmt): new connection from 10.0.16.10:46504
[   397.221330s]  INFO(runtime::session): new connection from 10.0.16.10:56988
[   397.227896s]  INFO(runtime::session): idle kernel interrupted
[   397.469244s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 32
[   398.123405s]  INFO(runtime::session): no connection, starting idle kernel
[   398.154819s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 32
[   399.095015s]  INFO(runtime::mgmt): new connection from 10.0.16.10:46510
```

jordens, 2018-03-09:
Might also be underlying #940.

sbourdeauducq, 2018-03-23:
Reproducible on the SYSU target. An idle kernel does not appear to be necessary; I can make the error appear by interrupting one regular kernel with another regular kernel (e.g. running ``artiq_run`` while another instance was already running).

sbourdeauducq, 2018-03-23:
KC705 is also affected.

sbourdeauducq, 2018-03-23:
I guess what happens is simple: there are events programmed far into the future by the first kernel (since the LED frequency is low). When the second kernel takes over, the value of ``now_mu`` given by ``break_realtime`` is the value of the RTIO counter plus some delta, which is less than the timestamp of some already programmed events.

We should be able to fix that either:
* by querying the RTIO core for the timestamp of the latest event written, and making sure that ``core.break_realtime()`` sets the timeline to at least this value. That timestamp should be reset with ``core.reset()``. There can be multiple (D)RTIO cores, so the firmware will have to iterate on them.
* by making it illegal (raise an exception) to call ``core.break_realtime()`` when a kernel has just been interrupted.

whitequark, 2018-03-23:
Or just maintaining a counter somewhere in the runtime that records the maximum timestamp seen.

sbourdeauducq, 2018-03-23:
No, the runtime is slow enough already, and this would not work with DMA.

jordens, 2018-03-23:
Pinning the timestamp (`now_mu`) to the timestamp CSR would also make it survive across kernel evictions/crashes.

sbourdeauducq, 2018-03-23:
The timestamp CSR does not necessarily contain the highest value submitted.

sbourdeauducq, 2018-03-23:
It might be good enough though (and simple and quite consistent with the current break_realtime behavior within a kernel).

jordens, 2018-03-23:
Yes. Pinning `now` to the CSR might also be a speed advantage (https://github.com/m-labs/artiq/issues/636). The only issue might be atomicity. But it would be an improvement in any case.

whitequark, 2018-03-23:
> No, the runtime is slow enough already, and this would not work with DMA.

This would work with DMA of course, since the kernel knows how long every DMA trace is already.

> Pinning now to the CSR might also be a speed advantage (#636).

Right now LLVM is free to eliminate consecutive loads and stores to `now` with no function calls in between, so this will have to be somehow preserved. I don't think it can do that for an integer cast to pointer. We might need to write a custom LLVM pass. And since the CSR is 64-bit, then every store to it will have to be followed by another store that commits the new value, so it looks like this will be strictly slower.

sbourdeauducq, 2018-03-23:
The commit part can be done by the gateware when one given 32-bit part of the 64-bit word is written. LLVM would have to know about this, though.

jordens, 2018-03-23:
Maybe just use the timestamp csr instead of the global and drop the kernel protocol to store/load now. That would be a start.

sbourdeauducq, 2018-12-01:
#636

closed 2018-12-01 by sbourdeauducq