m-labs/artiq #951: Sayma: assorted vivado warnings

hartytp, 2018-03-09:
A few things from the vivado output. Not sure if any are actually relevant/a cause for concern. Would be great if someone with more experience (@enjoy-digital, @sbourdeauducq or @jordens) could go through the vivado outputs in more detail and see if they can spot anything interesting...

AMC:

``` WARNING: [Place 30-568] A LUT 'IDDRE1_i_1' is driving clock pin of 5 registers. This could lead to large hold time violations. First few involved registers are:```
	```IDDRE1_4 {ISERDESE3}
	IDDRE1_3 {ISERDESE3}
	IDDRE1_2 {ISERDESE3}
	IDDRE1_1 {ISERDESE3}
	IDDRE1 {ISERDESE3} ```

```WARNING: [Vivado 12-3521] Clock specified in more than one group: serwb_pll_pll_serwb_serdes_clk [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:743]```

```WARNING: [Vivado 12-3521] Clock specified in more than one group: serwb_pll_pll_serwb_serdes_5x_clk [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:745]```

```CRITICAL WARNING: [Common 17-165] Too many positional options when parsing 'main_bufgce_div/O', please type 'get_nets -help' for usage info. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:695]```

```CRITICAL WARNING: [Common 17-165] Too many positional options when parsing 'main_bufgce/O', please type 'get_nets -help' for usage info. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:697]```

```WARNING: [Synth 8-6040] Register { driving address of a ROM cannot be packed in BRAM/URAM because of presence of initial value.```

```WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:474]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:475]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:479]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:480]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:484]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:485]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:489]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:490]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:494]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:495]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:499]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:500]
WARNING: [Vivado 12-4702] SLEW is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:534]
WARNING: [Vivado 12-4702] DRIVE is not a supported property on input port(s). Setting is ignored. [/home/ion/scratch/tph/artiq/artiq_sayma/standalone/gateware/top.xdc:535]```
RTM:

```"WARNING: [DRC PDRC-153] Gated clock check: Net CLKB0 is a gated clock net sourced by a combinational pin ISERDESE2_i_1/O, cell ISERDESE2_i_1. This is not good design practice and will likely impact performance. For SLICE registers, for example, use the CE pin to control the loading of data.```

```WARNING: [DRC PLHOLDVIO-2] Non-Optimal connections which could lead to hold violations: A LUT ISERDESE2_i_1 is driving clock pin of 1 cells. This could lead to large hold time violations.```

sbourdeauducq, 2018-03-09:
> CRITICAL WARNING: [Common 17-165] Too many positional options when parsing 'main_bufgce_div/O', please type 'get_nets -help' for usage info. [...]

Ah. I was surprised that the examples in https://www.xilinx.com/support/answers/67885.html worked correctly - of course they did not. Somehow I missed those warnings. Will check.

The SLEW and DRIVE ones are harmless.

hartytp, 2018-03-09:
Is it worth adding some basic scraping to the build scripts to flag up critical warnings and timing violations?

sbourdeauducq, 2018-03-09:
Seems it should be ``-of_objects`` and not ``of`` contrary to what the AR says.

marmeladapk, 2018-03-09:
@hartytp There's myriad of warnings and critical warnings you can get. If we could for example suppress some levels of information they would stand out more.

whitequark, 2018-03-09:
@hartytp We already flag the build on timing violations (it's a warning in buildbot). Not for critical warnings because there are so many for bullshit reasons, like adding files with multiple TCL commands and not one...

hartytp, 2018-03-09:
@whitequark Sure, but it's not hard to do some basic filtering to remove the more BS ones.

Anyway, in the meantime, I'd really appreciate it if someone who knows what to look for could go over all the vivado outputs carefully and see if they can spot anything I'd miss. It's tedious and time consuming, but worthwhile given how much of everyone's time this issue has consumed...

sbourdeauducq, 2018-03-09:
It seems to be ``-of``, which is not documented but appears in other examples and application notes. Vivado no longer complains about the command.

hartytp, 2018-03-09:
Also:
```WARNING: [Power 33-332] Found switching activity that implies high-fanout reset nets being asserted for excessive periods of time which may result in inaccurate power analysis.```

dtcallcock, 2018-03-09:
> high-fanout reset nets being asserted for excessive periods of time which may result in inaccurate power analysis

You must construct additional pylons!

hartytp, 2018-03-10:
@sbourdeauducq The critical warnings have now gone, as have the timing violations. Thanks!

If none of the other vivado warnings here concern you then feel free to close the issue.

sbourdeauducq, 2018-03-10:
> WARNING: [Synth 8-6040] Register { driving address of a ROM cannot be packed in BRAM/URAM because of presence of initial value.

That one is somewhat concerning - the name (``{``) of the register is changing at every run and is often binary garbage arising from Vivado memory corruption...

hartytp, 2018-03-10:
Hmm...any suggestions how we track that down? Does this only appear on Sayma? Worth a bit of old fashioned comment stuff out until the error goes so we can localize it? Suggestions about where to begin?

sbourdeauducq, 2018-03-10:
Maybe @jbqubit's Xilinx contacts can look into that one, since this is clearly a Vivado bug.
It doesn't occur only on Sayma, I know someone who also has this bug with a completely different FPGA project.

hartytp, 2018-03-10:
The garbled output is clearly a vibado bug, but isn't it still pointing at a real gateware issue that we should fix?

sbourdeauducq, 2018-03-10:
That looks like a minor detail that is sub-optimal; as long as Vivado produces a correct netlist (despite the memory corruption) and it passes timing then it's fine.

hartytp, 2018-03-10:
Or is the apparent memory corruption in vivado the only thing that worries you about that message? If so then I agree there isn't much we can do about it.

hartytp, 2018-03-10:
Okay. Well if there isn't anything else in the vivado output for sayma  which we need to look at further then feel free to close this.

sbourdeauducq, 2018-03-10:
> Or is the apparent memory corruption in vivado the only thing that worries you about that message?

Yes.
Well there's that, and this one on the RTM should be fixed:

> "WARNING: [DRC PDRC-153] Gated clock check: Net CLKB0 is a gated clock net sourced by a combinational pin ISERDESE2_i_1/O, cell ISERDESE2_i_1. This is not good design practice and will likely impact [...]

hartytp, 2018-03-10:
My thinking here is that the sdram issue could be vivado doing sonethibg silly in response to struggling to neet timing on the complex sawg desing. So, tripple checking all the cobstraints and fixing issues like this whuch are upsetting vivado could be a big help even if the warnings aren't directly related to the sdram.

sbourdeauducq, 2018-03-22:
> RTM:
>
> "WARNING: [DRC PDRC-153] Gated clock check: Net CLKB0 is a gated clock net sourced by a combinational pin ISERDESE2_i_1/O, cell ISERDESE2_i_1. This is not good design practice and will likely [...]

I don't see those (nor do I see something in the code that could cause them - we also use that local CLKB inversion for the SDRAM PHY on Kasli and this is the correct way to do it AFAIK). @hartytp what Vivado version are you using? 2017.4 here.

hartytp, 2018-03-23:
2017.4 here, although I've subsequently upgraded to 2017.4.1

I haven't rebuilt for a while so those messages may no longer be there.

Once I get my board back from @enjoy-digital I'll rebuild and look at what warnings remain.

sbourdeauducq, 2018-03-23:
According to Xilinx the .1 does not make a difference for Ultrascale.

sbourdeauducq, 2018-03-23:
(the one we have at least)

sbourdeauducq, 2018-03-27:
https://github.com/m-labs/artiq/issues/970 - reporting to Xilinx.
Other messages are fixed or not a cause for concern.

closed 2018-03-27 by sbourdeauducq