m-labs/artiq #953: "No experiments in module" when run  'artiq_run device_db.py'

mingshenli, 2018-03-11:
run  'artiq_run device_db.py'
[image]
I wonder if there are some problem in the file:
[image]
I only changed the core_addr.  I wonder if there are any other things to change here.

mingshenli, 2018-03-11:
and the only parameter I should change is the core_addr if I want to run the `led.py` using `artiq_run`?

jordens, 2018-03-11:
Depends. The `device_db.py` is correct if it corresponds to your firmware, gateware, and hardware. IP address is only one parameter there. Are you really using the KC705 with the NIST backplane and the AD9914 DDSs (since that's where you copied that device database from)?

mingshenli, 2018-03-11:
No,we only have a kc705 board.And now the board can not be found, that may also due to the `device_db.py`
[image]
We have run `artiq_flash -t [board] -m [adapter] -f flash_storage.img proxy storage start` containing the IP adderss successfully.

jordens, 2018-03-11:
Can you ping the board?
What are the messages on the serial port?

mingshenli, 2018-03-12:
I ping the IP address and it reply 'Unable to access the target host '

closed 2018-03-13 by mingshenli