m-labs/artiq #956: the kc705 board can not be found through Internet

mingshenli, 2018-03-13:
[image]
and when I ping the IP address, it reply that 'Unable to access the target host '
I have set the IP address using `artiq_mkfs flash_storage.img -s mac 00:0A:35:03:33:45 -s ip 192.168.1.170
`and ` artiq_flash -t kc705 -m nist_clock -f flash_storage.img proxy storage start` successfully.

cjbe, 2018-03-13:
@mingshenli you can debug this by checking the following:
* The network parameters are correct, i.e. the KC705 is on the same subnet as your computer
* Your computer is connected to the KC705 by a switch / hub that you manage (i.e. not someone elses switch that is filtering out your traffic)
* The KC705 is booting properly, and the configuration is correct - use flterm on the appropriate USB port

mingshenli, 2018-03-13:
the IP address is corect and hub is fine too.  What is 'use flterm on the appropriate USB port' means and how to make sure that 'The KC705 is booting properly, and the configuration is correct '?

cjbe, 2018-03-13:
If you run `python -m serial.tools.list_ports -v` you should see an entry `CP2103 USB to UART Bridge Controller` - this is the KC705 serial terminal. If you don't see this, make sure you have a USB mini cable connected to the UART port on the KC705, and that the driver is installed (https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers)

For me, on Linux, it is `/dev/ttyUSB1`, if you are on windows, it will be `COM<number>`
On Linux you can use `flterm` to monitor the serial port output. On Windows you can use Putty or Realterm - use a baud rate of 115200.

After you have the serial port connected, restart the KC705 - you can do this by pressing the button SW14 on the KC705, or by running `artiq_flash -t kc705 start`. If you have it right, within 10 seconds you should see a page of startup messages, including a message like
```
[     4.085562s]  INFO(runtime): using MAC address 00-30-d3-0e-af-ae
[     4.091527s]  INFO(runtime): using IP address 10.255.6.158
```
you should make sure that the IP and MAC address here are correct.

mingshenli, 2018-03-13:
ok,I will have a try, thanks.

sbourdeauducq, 2018-03-15:
Went there and fixed it.

closed 2018-03-15 by sbourdeauducq