m-labs/artiq #957: Kasli v1.1 I2C error on reset

cjbe, 2018-03-13:
Using a Kasli v1.1 with a derivative of DRTIO satellite gateware from 2edf65f57b86fa9797d5f9cb300191cec31e50ba, I see the following error every 1 in ~10 resets:

```
[     0.017152s] ERROR(board_artiq::i2c::imp): SDA is stuck low on bus #0
panic at src/libcore/result.rs:906: cannot initialize Si5324: "PCA9548 failed to ack control word"
```

whitequark, 2018-03-17:
I've did some investigation. Reinitializing Si5324 in a loop never results in stuck SDA. Neither does rebooting the board in a loop when done at safe points, i.e. when there is no I2C communication. As far as I can tell, what's happening is, when the I2C bus is not properly reset, such as if abruptly resetting the core device, the slave doesn't see any SCL pulses and holds SDA low.

PCA9548 has a feature to deal specifically with this:
> An active LOW reset input allows the PCA9548A to recover from a situation where one of
the downstream I2C-buses is stuck in a LOW state.

Unfortunately, the PCA9548 reset is only wired to the FTDI chip on Kasli and not to the FPGA.

@sbourdeauducq, what is the process for getting a hardware change on Kasli?

jordens, 2018-03-17:
Try doing a bunch of SCL cycles (with SDA deasserted) as initialization. Unless there is a hardware issue pulling SDA/SCL this will recover it.

sbourdeauducq, 2018-03-17:
We have those boards out there right now, so whatever technique is implemented has to work without the hardware reset. Are there any ill effects from using SCL pulses when SDA is stuck instead of a dedicated reset anyway? We are initializing the I2C switches and resetting the Si5324 afterwards.

sbourdeauducq, 2018-03-17:
The dedicated hardware reset would also be partial anyway unless we connect a reset line to every I2C slave.

whitequark, 2018-03-17:
That's a good point, there's no room on the extension connectors for that anyway.

gkasprow, 2018-03-17:
There are pin-compatible MAXIM [chips](https://www.maximintegrated.com/en/products/analog/analog-switches-multiplexers/MAX7358.html) that have bus stuck detection which causes disconnection of problematic branch

commit c86df8e1 (2018-03-19): firmware: try to unstuck the I2C bus if it gets stuck.

Fixes #957.
closed 2018-03-19 by whitequark