m-labs/artiq #958: DRTIO Master-Satellite fine timing alignment on Kasli

cjbe, 2018-03-13:
Using a Kasli master and satellite based off 2edf65f57b86fa9797d5f9cb300191cec31e50ba, I see that the timing of SERDES outputs on the satellite relative to the master changes over restarts across a window of 8ns.

The phase relationship of the master and satellite si5234 outputs stays the same over restarts.

If I replace the serdes outputs with ttl_simple outputs, the timing alignment does *not* change over restarts.

sbourdeauducq, 2018-03-22:
@cjbe Does it work correctly now?

cjbe, 2018-03-22:
I tested ce2b5a97cba06cb3c4b1d869ae116fbe5702a934 - this did not fix the issue.
I could not see any relevant commits since then.

sbourdeauducq, 2018-03-22:
I suppose that you tested between a SERDES TTL on the master and a SERDES TTL on the satellite, correct?

cjbe, 2018-03-22:
@sbourdeauducq correct

sbourdeauducq, 2018-03-22:
The satellite was resetting the SERDES, but the master was not (this may also cause problems on non-DRTIO targets). Hopefully fixed now!

cjbe, 2018-03-23:
@sbourdeauducq this fixes it. I observe no change in timing alignment between master and satellite serdes outputs over 30 restarts.

closed 2018-03-23 by cjbe