m-labs/artiq #961: o1rk-linux-ld cannot open output file

JKiethe, 2018-03-15:
directory exists
user can write to directory

```
root:Terminating with exception (Exception: or1k-linux-ld invocation failed: or1k-linux-ld: cannot open output file C:\Users\MULTI-~1\AppData\Local\Temp\tmph_pp0d7o: Permission denied)
Traceback (most recent call last):
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\master\worker_impl.py", line 273, in main
    exp_inst.run()
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\language\core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\coredevice\core.py", line 128, in run
    self.compile(function, args, kwargs, set_result)
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\coredevice\core.py", line 111, in compile
    library = target.compile_and_link([module])
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\compiler\targets.py", line 180, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\compiler\targets.py", line 170, in link
    **{"obj{}".format(index): obj for index, obj in enumerate(objects)}) \
  File "C:\Users\multi-ion\AppData\Local\conda\conda\envs\artiq-4-dev\lib\site-packages\artiq\compiler\targets.py", line 37, in __enter__
    format(cmdline[0], stderr.decode('utf-8')))
Exception: or1k-linux-ld invocation failed: or1k-linux-ld: cannot open output file C:\Users\MULTI-~1\AppData\Local\Temp\tmph_pp0d7o: Permission denied
```

JKiethe, 2018-03-15:
(filed for @jordens)
@whitequark any idea?

jordens, 2018-03-15:
Potentially #935

whitequark, 2018-03-15:
@jordens Unlikely, this is probably caused by the 8.3 path: `C:\Users\MULTI-~1\...`

commit 4b5a78e2 (2018-03-15): compiler: do not pass files to external tools while they are opened.

This fixes access violations on Windows that are present both with
input and output files. For some reason, Cygwin-compiled binutils
did not exhibit this problem, but MSYS-compiled binutils do.

Fixes #961.
commit ba017147 (2018-03-15): compiler: do not pass files to external tools while they are opened.

This fixes access violations on Windows that are present both with
input and output files. For some reason, Cygwin-compiled binutils
did not exhibit this problem, but MSYS-compiled binutils do.

Fixes #961.
closed 2018-03-15 by whitequark