m-labs/artiq #965: Backup for artiq_dashboard.pyon file

r-srinivas, 2018-03-20:
We are using artiq 2 on Windows and recently encountered a gui crash that made our artiq_dashboard.pyon file blank. That's probably a separate issue that I am still trying to replicate but it resulted in all our information about the gui such as positions of docks and more importantly applet information to be erased. 

Fortunately, we had a backup from the previous day so it didn't really cost us. However, since the file isn't in the repository and is in some other folder as specified in the release notes, 

>GUI state files for the browser and the dashboard are stores in “standard” locations for each operating system. Those are ~/.config/artiq/2/artiq_*.pyon on Linux and C:\Users\<username>\AppData\Local\m-labs\artiq\2\artiq_*.pyon on Windows 7.

This might be a problem for someone else that encounters this as this would not normally be backed up. I'm not sure if there's a good way of generating a backup, but maybe automatically saving the pyon file from the previous day, or the option to make a backup from a known safe state would be useful.

jordens, 2018-03-20:
Writing the new state file should be atomic. And when a file can't be read (but exists) it should not be cleared/overwritten. AFAICT that's the case.

https://github.com/m-labs/artiq/blob/9ad1fd8f25328c35a92e67ceac95c93d6d3dddab/artiq/gui/state.py#L71

https://github.com/m-labs/artiq/blob/9ad1fd8f25328c35a92e67ceac95c93d6d3dddab/artiq/protocols/pyon.py#L213

Unrelated: remind me why you can't upgrade to 3.x.

r-srinivas, 2018-03-20:
> Writing the new state file should be atomic. And when a file can't be read (but exists) it should not be cleared/overwritten. AFAICT that's the case.

The file got overwritten with a blank file. I forget the error message but it was something along the lines of not being to load a pyon file with null bytes.

> Unrelated: remind me why you can't upgrade to 3.x.

We've been trying to finish some experiments and don't really want to refactor code to do the upgrade currently.

jordens, 2018-03-20:
I can't see how that could happen given the code. Unless there are hard disk problems or windows was particularly unhelpful. Or if the error causing the generation of a new cleared settings file is somewhere else. But maybe there is some feature that could be added to recover from that.
Also applets should be defined using `ccb`.

Ack. Please let us know if there are other issues preventing you from upgrading.

whitequark, 2018-03-30:
@jordens According to a Microsoft employee Windows provides no guarantees of atomicity for `MoveFile`/`MoveFileEx` (what Python uses for `os.replace` on Windows), see [this](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/449bb49d-8acc-48dc-a46f-0760ceddbfc3/movefileexmovefilereplaceexisting-ntfs-same-volume-atomic?forum=windowssdk#a239bc26-eaf0-4920-9f21-440bd2be9cc8).

jordens, 2018-08-07:
I have seen this (the state being NULL bytes) as well after a blue screen on windows. It would be great if we could work around windows file system limitations by just keeping one previous version of the file around, e.g. the one that was last successfully used to restore the state on dashboard/browser start.