m-labs/artiq #966: inconsistent serwb behavior when RTM FPGA is not loaded

sbourdeauducq, 2018-03-21:
Sometimes when the RTM FPGA is not loaded and the AMC starts, it prints several times:
```
[     1.454088s]  WARN(board_artiq::serwb): AMC/RTM serwb bridge initialization failed, retrying.
```
and sometimes it just waits for it to be loaded. Why are there those two different behaviors at random?

sbourdeauducq, 2018-04-30:
@enjoy-digital This is still a problem with the current master; can you look into this?

enjoy-digital, 2018-04-30:
Understood and fixed via https://github.com/m-labs/artiq/commit/64c8eee28d6eec7db071842c8a67d21880998142.

sbourdeauducq, 2018-05-10:
This is still happening.

enjoy-digital, 2018-05-11:
When RTM is not loaded, the AMC LVDS input is not driven and can be 0 or 1. The case is now handled in the gateware with https://github.com/m-labs/artiq/commit/cd4477864ae9f47d6535ef8d59ab817c151f9abc.

If input starts toggling, we need to start calibration since this means RTM is calibrated and is sending pattern to AMC for calibration.

Once https://github.com/m-labs/artiq/issues/813 will be implemented, serwb initialization will only be done after RTM is loaded. So this is a false problem: It will not happen and even if it happens, it falls back correctly (retry link calibration).

closed 2018-05-11 by enjoy-digital