m-labs/artiq #967: fix/simplify serwb clocking

sbourdeauducq, 2018-03-21:
The serwb clocking is needlessly complicated (too many clock domains) and precludes the use of Xilinx-recommended clocking schemes for the IOSERDES:
https://www.xilinx.com/support/documentation/application_notes/xapp1324-design-selectio-component-primitives.pdf
https://www.xilinx.com/support/answers/67885.html

I suggest the following:
* move all clocking (MMCMs etc.) out of serwb, make them part of the system CRG.
* operate at 1Gbps (there is already a gearbox for non-integer ratios, so that should not be very complicated?).
* reuse existing SDRAM and system clocks.

Note: for speed grade 1 (which is what we have), on HR banks, the SERDES in less-insane (as opposed to "native") mode is spec'd to 1000Mbps only.

hartytp, 2018-03-26:
What's the timeline for this? If the mem test now works, the SER WB is the main source of unreliability on Sayma from the ARTIQ side of things AFAICT. It would be great to have this working more robustly.

enjoy-digital, 2018-03-26:
@hartytp: i'll work on that on wednesday.

enjoy-digital, 2018-03-29:
@sbourdeauducq: i just want to check things with you before implementing:
- serwb_serdes_20x_clk would be sys4x_clk
- serwb_serdes_5x_clk would be sys_clk
- serwb_serdes would be removed, we would use sys_clk with stb/ack dataflow control.

If so, should i keep specific serwb_serdes clock domains (ie serwb_serdes_4x/serwb_serdes) or should i use sys4x_clk/sys_clk?

sbourdeauducq, 2018-03-29:
Use sys4x_clk/sys_clk - the Wishbone side is in the sys_clk domain and synchronous to the SERDES.

hartytp, 2018-04-02:
What's the status of this? It's been a while since the SDRAM was fixed, and I'd love to have a version of Sayma I can actually use.

enjoy-digital, 2018-04-02:
@hartytp: yes sorry, i started working on that but need to test and commit.

hartytp, 2018-04-03:
@sbourdeauducq With the caveat that I'm not aware of the precise details of any of the relevant contracts, and with apologies if this is treading on peoples' toes here, but...

If @enjoy-digital isn't able to complete this work in a reasonably timely fashion, can you take it over, please?

The impression I get is still that @enjoy-digital isn't able to put serious time into getting Sayma to work (cf how long we wasted on the SDRAM issues because no one had looked up the relevant Xilinx guides and implemented the recommendations until @jordens stepped in) and that this is going to drag on until someone actually takes ownership of the project puts proper time into it.

enjoy-digital, 2018-04-03:
Here is what i did:
https://github.com/enjoy-digital/artiq/commit/2e61649ba230aefd788dced5cce21bf1fcc2e5c4
https://github.com/enjoy-digital/artiq/commit/3c02cc5f9eea901d6d58bef88b9e00b2b218ef42
https://github.com/enjoy-digital/misoc/commit/4b460283d6ec2335436c71241e868a6bf42ece3a

I was doing some tests with a simple design but crashed JTAG on sayma3 with CTRL-C...

@sbourdeauducq: can you integrate it and test it?

sbourdeauducq, 2018-04-03:
> I was doing some tests with a simple design but crashed JTAG on sayma3 with CTRL-C...

Doesn't @whitequark's usbreset.c work around this bug?

enjoy-digital, 2018-04-03:
@sbourdeauducq: i'm going to try, can you review the code to tell me if that's what you want?

sbourdeauducq, 2018-04-03:
No, that's not what was described here: https://github.com/m-labs/artiq/issues/967#issuecomment-377282994
We should not need a 25MHz clock, or any additional clock for that matter.

What I was thinking was:
* the SERDES is in 8:1 mode and operates at 1Gbps line rate.
* a "gearbox", running at 125MHz, takes 8-bit words from the ISERDES and turns them into 10-bit words (with a strobe signal indicating valid words) for subsequent 8b10b decoding.
* another gearbox (with an ack signal indicating word acceptance) does 10->8 for the OSERDES.
* I don't think using the dataflow stream classes is necessary there since it's just a strobe signal in one case and a ack signal in the other (but that's a detail, the stream classes are OK if you really prefer them).

sbourdeauducq, 2018-04-03:
Do we have a HP or HR bank?
HR is limited to 1.00Gbps. If we have HP, maybe we can instead keep 1.25Gbps, replace that stupid Ultrascale IOSERDES with IODDR (from my experience with RGMII, that one component wasn't lousy with bugs and misfeatures, though there may be surprises at high speeds), and do 5:10 in fabric.
Or even drop the special IOs completely and operate at a speed attainable by the registers (>500Mbps should still be attainable I believe, though the IODELAY range may then become an issue).

enjoy-digital, 2018-04-03:
I've implemented something and will test it tomorrow.

enjoy-digital, 2018-04-04:
I still have to do some debug, but i should not be that far from having something working using only sys/sys4x clock.

enjoy-digital, 2018-04-06:
It's working on a simple design (Nexys Video / SERWB overa HDMI cable and the two HDMI TX/RX ports / A master and a Slave in the same design). I'm doing more testing on that (P&R is only 30 seconds, so it's easier), cleaning up things, and i'll test/integrate in ARTIQ.

enjoy-digital, 2018-04-07:
Integrated and tested on ARTIQ. Scrambling is not enabled yet. I'll try to enable it next week. That would be good if @hartytp (or someone else) could do more testing and give me some feedback now that it's fresh in my mind.

hartytp, 2018-04-07:
Great. Will look at that soon

hartytp, 2018-04-09:
@enjoy-digital Thanks for doing that!

I'll have a look at that now. Anything you want me to do beyond power cycling it a few times and looking for errors?

enjoy-digital, 2018-04-09:
@hartytp: yes you can do that. You also said you had spi issues with trying to debug hmc830. If that's something easy to reproduice, that would be interesting to test you still have issues.

hartytp, 2018-04-09:
I never got to the point of having a proper test that reproducibly crashed the ser-WB as there were other more pressing issues at the time.

I'll start using Sayma again and keep an eye out, it things are as bad as before I'll see it soon enough.

enjoy-digital, 2018-04-09:
ok fine, thanks.

hartytp, 2018-04-13:
Here is as many repeats as I could be bothered with of: `artiq_flash -t sayma start` followed by `openocd -f sayma_new.cfg -c "pld load 0 rtm.bit; exit"` https://hastebin.com/soxibaweka.sql
- Each time, I see `[     0.028287s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...` `[     0.574446s]  INFO(board_artiq::serwb): done.` as expected.
- I did not see any occurrences of #856 or #966

hartytp, 2018-04-13:
Thanks @enjoy-digital !

sbourdeauducq, 2018-04-13:
@hartytp And does it go further (e.g talks to the HMC7043), or crashes somewhere?

hartytp, 2018-04-13:
No, it just sits at `[     7.533404s]  INFO(board_artiq::serwb): done.`

Remind me what it should do? Not sure if having the 2V/4V supplies disabled is the issue here?

FWIW, I see exactly the same behavior with power cycling Sayma instead of using artiq_flash

sbourdeauducq, 2018-04-13:
Then something is broken; it should at least print the RTM FPGA gateware version.

hartytp, 2018-04-13:
I'm not up to speed enough on the hw/gw/sw to know whether this is expected when these two supply rails are taken down.

sbourdeauducq, 2018-04-13:
This is not expected. If serwb initializes correctly then the version should be printed; this does not require additional RTM FPGA voltages.

hartytp, 2018-04-13:
Okay, thanks for confirming.

sbourdeauducq, 2018-04-13:
Problem is not present on the HKG sayma-3.
```
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003886s]  INFO(runtime): software version 4.0.dev+823.gf96f597e
[     0.010153s]  INFO(runtime): gateware version 4.0.dev+823.gf96f597e
...
[     0.028286s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.200124s]  INFO(board_artiq::serwb): done.
[     0.203213s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+823.gf96f597e
```
Works consistently after several restarts.

hartytp, 2018-04-13:
@gkasprow Do you want me to just send you my board back?

gkasprow, 2018-04-13:
Can I hold it for a while. I want to make sure that Ethernet works. I have 2 AMC boards and on each of them Ethernet PHY works in different manner :)

hartytp, 2018-04-13:
@gkasprow Fine, I'll put this on hold until the ethernet is fixed. Let's track these bugs down properly one at a time.

jbqubit, 2018-04-13:
I built from master today using ```--without-sawg```. serwb is successful over 10 reboots of Sayma (s/n 8). Great work @enjoy-digital!

Same as @hartytp I see no sign of RTM gateware. 

```
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+796.g5ca59467.dirty
Initializing SDRAM...
DQS initial delay: 111 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010010000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010010000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 111 taps
Write leveling: 109 104 120 115 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000110101010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010101011101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 229+-85 213+-94 197+-89 187+-87 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
[     0.010157s]  INFO(runtime): gateware version 4.0.dev+796.g5ca59467.dirty
[     0.016980s]  INFO(runtime): log level set to INFO by default
[     0.022677s]  INFO(runtime): UART log level set to INFO by default
[     0.028809s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.418596s]  INFO(board_artiq::serwb): done.
```

jbqubit, 2018-04-13:
Continue test using same build of master with --without-sawg on Sayma (s/n 7). I am successfully loading RTM gateware. serwb is successful over 10 rounds of issuing ```artiq_flash -t sayma -f sayma.config storage start```. This board also does not show ```RTM gateware version...```. 

Looking into this more... 
- I unplug then plug-in USB and reload AMC gateware, it does advance to the point where ```RTM gateware version``` is printed. 
- Subsequent issuances of ```artiq_flash -t sayma -f sayma.config storage start``` don't print ```RTM gateware version```. 
- Without touching USB cable, I reload AMC gateware, it does advance to the point where ```RTM gateware version``` is printed. 
- Subsequent issuances of ```artiq_flash -t sayma -f sayma.config storage start``` don't print ```RTM gateware version```.

enjoy-digital, 2018-04-13:
@jbqubit: there is something wrong if you do not see the version. I'll have to look at that.

hartytp, 2018-04-13:
@enjoy-digital Thanks! Let me know if you need my amc + RTM to take another holiday in France (I'm quite jealous actually!).

sbourdeauducq, 2018-04-13:
@whitequark can you reconnect the other sayma to the server?

sbourdeauducq, 2018-04-13:
@enjoy-digital Are you implementing all the tricks that were found to be necessary for the Ultrascale IO garbage to behave with SDRAM into serwb?

enjoy-digital, 2018-04-14:
@hartytp, @jbqubit: sorry, stupid question, but was the rtm gateware also regenerated? (i just want to be sure)

Note that with old serwb, we were not able to run it a 1.25Gbps on all boards and reduced it to 625Mbps. It's now increased to 1Gbps, so it would be interesting to test on the others sayma as @sbourdeauducq is suggesting since IIRC it was not working correctly at 1.25Gbps.

@sbourdeauducq: we should implement the tricks but i'll double check.

hartytp, 2018-04-14:
@enjoy_digital yes I rebut everything from scratch

hartytp, 2018-04-14:
What about adding a prbs check to verify the link/clocking?

enjoy-digital, 2018-04-14:
hartytp: yes that's a good idea. I'll add a prbs check after initialization.

hartytp, 2018-04-14:
Great, thanks!

sbourdeauducq, 2018-04-15:
You should be able to go to 500Mbps, and keep the same clocks, by putting the SERDES into 4:1 mode.
Speed for serwb is not very important, the priority is to get something to work at all and without intermittent or board-dependent bugs that waste the maximum amount of time.

sbourdeauducq, 2018-04-15:
Though, the maximum range of IDELAY varies between 1.28ns and 7.68ns with PVT (not a typo, welcome to Ultrascale). So, anything below ~780Mbps has to deal with the IDELAY range being maxed out, otherwise we will stay in the above-mentioned bug hell. Maybe the OSERDES can be in 4:1 and the ISERDES in 1:8 and you switch to the other set of outputs when the IDELAY maxes out. Use EN_VTC, I don't trust the Ultrascale taps to be stable over any significant amount of time.

enjoy-digital, 2018-04-16:
I've reduced serwb speed to 500Mbps, it will maybe improve things.

@hartytp:  I've added PRBS on my simple design, but if we want to add PRBS to serwb in ARTIQ i have to think a bit how things should sequenced, since we are not able to control the RTM when in PRBS mode.
@jbqubit: your analysis is interesting, but i'm not sure to understand what's the difference between doing "artiq_flash -t sayma -f sayma.config storage start" and reloading the AMC, can you explain?

jbqubit, 2018-04-16:
By "reloading the AMC" I mean that I bypass the Flash by running [this](https://github.com/jbqubit/sinara-testing/blob/master/sayma/tools/load_amcrtm.bash).

sbourdeauducq, 2018-04-17:
> I've reduced serwb speed to 500Mbps, it will maybe improve things.

Not this way: the timing windows are still the same, you depend on the data being correct on an arbitrary hardcoded set of downsampled ISERDES bits.
Instead, you need to switch between two sets when the IDELAY maxes out.

sbourdeauducq, 2018-04-17:
> have to think a bit how things should sequenced, since we are not able to control the RTM when in PRBS mode.

Have the gateware send/receive exactly X megabytes of PRBS data then go back to normal operation?

sbourdeauducq, 2018-04-17:
Also I don't see how the scan algorithm can possibly work correctly right now, the delay range is less than 1 UI.
I suggest keeping 1Gb unless there is a clear reason not to, handling the delay range limit without bug is not straightforward.

enjoy-digital, 2018-04-17:
@sbourdeauducq: ok, i reverted linerate, moved some things, added a line test after initialization.
@jbqubit: i did some changes on RTM reset and added some debugs informations, can you do a test on your board?

jbqubit, 2018-04-17:
@enjoy-digital I'm building --without-sawg now and will test shortly.

enjoy-digital, 2018-04-17:
@jbqubit: thanks

jbqubit, 2018-04-17:
I built from master with --without-sawg. Using AMC+RTM. @enjoy-digital Looks like this was a step backward. memtest fails 10 of 10 times I tried on this board pair (AMC s/n -8). 

```
  __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+838.gfe689ab4.dirty
Initializing SDRAM...
DQS initial delay: 94 taps
Write leveling scan:
Module 3:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011111101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011010101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011011101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
0000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 94 taps
Write leveling: 80 85 106 107 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: Zero window: 3: 0-0 (0)
SDRAM initialization failed
Halting.
```

memtest passes on this same board (AMC s/n -8) for 20180413 build from master using --without-sawg. I checked again today with the 20180413 version and see that it works. 

---------------

Try with second AMC+RTM that's in the lab (AMC s/n -7). ... memtest and serwb passes but AD9154 init fails 6 of 6 times I tried. 

```
__  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+838.gfe689ab4.dirty
Initializing SDRAM...
DQS initial delay: 114 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111 [... cut ...]
[     4.575011s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #3 failed (AD9154 SERDES PLL lock timeout), retrying
[     4.594585s]  INFO(board_artiq::ad9154): AD9154-0 found
[     4.598501s]  INFO(board_artiq::ad9154): AD9154-0 configuration...
[     4.806013s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #4 failed (AD9154 SERDES PLL lock timeout), retrying
[     4.825587s]  INFO(board_artiq::ad9154): AD9154-0 found
[     4.829503s]  INFO(board_artiq::ad9154): AD9154-0 configuration...

```
memtest passes on this same board (AMC s/n -7) for 20180413 build from master using --without-sawg.

sbourdeauducq, 2018-04-17:
Please open two new issues for those two problems.

sbourdeauducq, 2018-04-17:
> [     0.003889s]  INFO(runtime): software version 4.0.dev+796.g5ca59467
> [     0.010157s]  INFO(runtime): gateware version 4.0.dev+838.gfe689ab4.dirty

And address the version mismatch before reporting any issues. Keep amc bitstream, rtm bitstream, bootloader and runtime in sync at all times, otherwise you are wasting time.

enjoy-digital, 2018-04-18:
@jbqubit: thanks for testing. You are indeed using old software, serwb should output more informations.

jbqubit, 2018-04-18:
@sbourdeauducq You're right, I didn't roll back ARTIQ when trying to revert back to my archived build from 20180413. The tests with 838.gfe689ab4.dirty were with matching version of ARTIQ.

jbqubit, 2018-04-19:
I built from master last night with SAWG. Here's what I see on my two Sayma AMC+RTM setup. Now with software gateware version match. 

# board 1

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+840.ga4f18770
Initializing SDRAM...
DQS initial delay: 98 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000100001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100111111111
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100111111111
Module 1:
000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101010111111111111111111111111
Module 0:
000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000100011011111111111111111111111111111
DQS initial delay: 98 taps
Write leveling: 57 57 78 74 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00101011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010101010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
11111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 97+-61 88+-67 68+-60 62+-63 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003888s]  INFO(runtime): software version 4.0.dev+840.ga4f18770
[     0.010154s]  INFO(runtime): gateware version 4.0.dev+840.ga4f18770
[     0.016456s]  INFO(runtime): log level set to INFO by default
[     0.022155s]  INFO(runtime): UART log level set to INFO by default
[     0.028288s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     2.456337s]  INFO(board_artiq::serwb): done.
[     2.459373s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     3.465020s]  INFO(board_artiq::serwb): 209716 errors
[     3.468768s]  INFO(board_artiq::serwb): AMC to RTM Link test

```

Running artiq_flash start several times in succession I see that the number of serwb errors is fixed at 209716.

# board 2

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited
[... cut ...]
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000006s]  INFO(runtime): ARTIQ runtime starting...
[     0.003888s]  INFO(runtime): software version 4.0.dev+840.ga4f18770
[     0.010154s]  INFO(runtime): gateware version 4.0.dev+840.ga4f18770
[     0.016456s]  INFO(runtime): log level set to INFO by default
[     0.022155s]  INFO(runtime): UART log level set to INFO by default
[     0.028288s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
```

[5 more replies]

closed 2018-04-30 by enjoy-digital