m-labs/artiq #970: Vivado outputs garbled message - memory corruption?

sbourdeauducq, 2018-03-27:
* Download http://m-labs.hk/sayma/vivado_6040.zip (this is the Verilog for Sayma AMC with SAWG)
* Run ``bash build_top.sh``
* Vivado produces the following message
```
WARNING: [Synth 8-6040] Register ���_ driving address of a ROM cannot be packed in BRAM/URAM because of presence of initial value.
```

The name of the register is obviously incorrect and varies between runs, which can be a sign of memory corruption that might have more serious effects.

sbourdeauducq, 2018-03-28:
Without mor1kx dependency: https://m-labs.hk/sayma/vivado_6040_nomor1kx.zip

sbourdeauducq, 2018-04-03:
According to Xilinx tech support:
"The messaging bug in Vivado 2017.4 only affects the message itself, and not the resulting netlist coming from it. The netlist will be correct. The issue is due to the message receiving the pointer for the actual name, rather than just the actual name of the register, which is why the message looks so garbled. As Marc mentioned, this is corrected in the upcoming 2018.1 Vivado release."

closed 2018-04-03 by sbourdeauducq

jbqubit, 2018-04-03:
Thanks for tracking this down @sbourdeauducq. Glad the conduit to Xilinx tech support is now open.