m-labs/artiq #975: not any files are installed when install artiq-3.6 version

joeshardow, 2018-04-01:
[image]
[image]
There are not any files when i install artiq 3.6. How can I build a normal environment?

whitequark, 2018-04-01:
You also need to install the `artiq` package.

joeshardow, 2018-04-02:
How to install artiq package now? The command "conda create -n artiq-main artiq-kc705-nist_clock"  is install everything before.

whitequark, 2018-04-02:
`activate artiq-main` then `conda install artiq`

sbourdeauducq, 2018-04-03:
@whitequark Is there a problem with the dependency of the board packages?

sbourdeauducq, 2018-04-03:
@whitequark Can be reproduced (on Linux); the board package no longer installs artiq.

commit 4e21a13b (2018-04-07): conda: put requirements on correct artiq-board package.

Fixes #975.
closed 2018-04-07 by whitequark