m-labs/artiq #976: Duplicate imports of experiments with identical module name

cjbe, 2018-04-03:
If an experiment repository contains modules with the same name, the experiments in all previously imported modules of the same name appear under the path for the currently imported module.

E.g. if scanning a repository:
```
a/x.py
b/x.py
c/x.py
```
in the repository explorer the folder `a` would contain the experiments from `a/x.py`, the `b` directory 
 would contain `a/x.py` and `b/x.py`, and `c` `a/x.py`, `b/x.py`, `c/x.py`.

sbourdeauducq, 2018-05-01:
@cjbe I cannot reproduce this; can you provide a tarball of a minimal experiment tree that causes the problem?

sbourdeauducq, 2018-05-01:
Ah nevermind, I reproduced it. My modules had the same experiment names in them, so the problem was masked.

commit b48e782d (2018-05-01): tools/file_import: restore sys.modules. Closes #976
commit 5f3e417b (2018-05-01): tools/file_import: restore sys.modules. Closes #976
commit 8d66d53e (2018-05-02): Revert "tools/file_import: restore sys.modules. Closes #976"

This reverts commit b48e782dd67282127a6bab96de5166a143e57f29.
commit b84a2a1e (2018-05-02): Revert "tools/file_import: restore sys.modules. Closes #976"

This reverts commit 5f3e417bb53b71680510d90e142009449565b350.
closed 2018-05-01 by sbourdeauducq