m-labs/artiq #977: the dashboard raise error

mingshenli, 2018-04-03:
I open the dashboard as the documentation  says, and successfully makes it print ''hello world''
[image]
however, when I create a file which hope the ttl windows on dashboard will react,(the file is in `~/artiq-master/repository` )
[image]
 open the dashboard, there is a error
[image]
and the dashboard
[image]
click submit, then

[image]
and the ttl windows on dashboard has no response.

and, on a VMware run linux Ubuntu，the dashboard can not be open

[image]

mingshenli, 2018-04-03:
ok. I will have a try. thanks

mingshenli, 2018-04-03:
no, now there is error
[image]
the`MgmtTutorial1` is the name it changed  from`rtio.py`, and `Mgmt_Tutorial.py`  is the name of the file I use to test the dashboard(print `hello world`)
and, can the dashboard function without the kc705 been connected through internet?

sbourdeauducq, 2018-04-03:
> heMgmtTutorial1 is the name it changed fromrtio.py, and Mgmt_Tutorial.py is the name of the file I use to test the dashboard(print hello world)

Right-click on the experiment list in the explorer and trigger a rescan.

> and, can the dashboard function without the kc705 been connected through internet?

Yes.

mingshenli, 2018-04-03:
it still raise the same error

sbourdeauducq, 2018-04-03:
Please post detailed steps explaining how I can reproduce the problem.

mingshenli, 2018-04-03:
after open the dashboard, and the cmd windows is same as upon pictures shows,
[image]

> Right-click on the experiment list in the explorer and trigger a rescan

I click `scan repository HEAD`
because it says `Renaming class 'MgmtTutorial' in 'rtio.py' to 'Management tutorial1'`, then I chose the 
Management tutorial1 and click submit, and then the cmd of artiq_master shows `ERROR:master:artiq.master.scheduler:got worker exception in prepare stage, deleting RID 49`

sbourdeauducq, 2018-04-03:
That's not enough for me to reproduce it, e.g. I don't have your experiments. I need detailed instructions that start from a *blank* ARTIQ installation.

mingshenli, 2018-04-03:
[image]
[image]

sbourdeauducq, 2018-04-03:
I see three experiments in your screenshot. Are they all needed to reproduce the problem? if yes then I need all three. If no then please delete the other ones and give me minimal instructions.

sbourdeauducq, 2018-04-03:
And please post the *text* of experiments, not a screenshot, so I can copy-paste it.

mingshenli, 2018-04-03:
the rtio.py:
from artiq.experiment import *

class Tutorial(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl0")

    @kernel
    def run(self):
        self.core.reset()
        self.ttl0.output()
        for i in range(1000000):
            delay(2000000*us)
            self.ttl0.pulse(2000000*us)

mingshenli, 2018-04-03:
the mgmt_tutorial.py
from artiq.experiment import *

class MgmtTutorial(EnvExperiment):
    """Management tutorial"""
    def build(self):
        pass  # no devices used

    def run(self):
        print("Hello World")

mingshenli, 2018-04-03:
and it is strange that I only put these two experiments in repository

mingshenli, 2018-04-03:
and, about the problem on ubuntu, I do as the guide says 'making a py35 environment' the same as I do on windows.
[image]

and we have pay the fee for Technical consultation when we buy the sinara

sbourdeauducq, 2018-04-03:
What is your email address?

mingshenli, 2018-04-03:
limsh9@mail2.sysu.edu.cn

mingshenli, 2018-04-03:
no. actually I make a python3.5 environment as the documentation says. should I create a python 3.6 environment use `conda create -n py36 python=3.6 anaconda`?

sbourdeauducq, 2018-04-03:
I meant **ARTIQ*** 3.6, not Python 3.6.

mingshenli, 2018-04-03:
no, I use artiq3.4. how to update? 
[image]
the order can not be used for a long time, says files can not be found.

sbourdeauducq, 2018-04-03:
``conda create -n artiq-new artiq=3.6 artiq-kc705-nist_clock=3.6``

mingshenli, 2018-04-03:
ok. after installing 3.6 as the order you give,  I will try it on a new version, hoping the new version will solve the problem. the installing will take a long time, I will let you know once I reproduce the step above.
thanks alot.

sbourdeauducq, 2018-04-03:
The software also behaves correctly when I make a copy of the experiment ``Management tutorial`` (renames one of them, they both can still run, and there is no bug when rescanning the repository).

sbourdeauducq, 2018-04-03:
Also reflash the board after upgrading.

mingshenli, 2018-04-03:
ok. hope the upgrading will solve the problem.
thanks.

sbourdeauducq, 2018-04-03:
It should fix the cygwin one, but nothing has changed regarding repository management, so I think the experiment name issue will persist.

mingshenli, 2018-04-04:
here is every move:
`activate artiq-new`;   `cd C:\Users\18926\AppData\Local\conda\conda\envs\artiq-new\artiq-master`
`artiq_master`;  and open a new cmd, activate the artiq-new, ` cd C:\Users\18926\AppData\Local\conda\conda\envs\artiq-new\artiq-master` ; `artiq_dashboard`

[image]
and then I click submit, the experiment is same as above,
all of the log since the dashboard been opened:(the chinese means ''error sem timeout''

[image]

[image]

sbourdeauducq, 2018-04-04:
So, it seems that the first problem you were reporting is resolved: There are no more incorrect names in the explorer, and the experiment starts when you submit it.

The two errors you now have are:
1. the dashboard failing to connect to the core device for moninj (i.e. TTL and DDS panels)
2. the experiment, once started, failing to connect to the core device.

Is the core device (the KC705 board) properly set up and connected to the network? Can you ping it? Did you set its IP address in ``device_db.py``?

mingshenli, 2018-04-04:
well, we didn't connect to the kc705(as you said it do not have to)
and the problem is ''get worker exception'', and, the ttl window has no respons

sbourdeauducq, 2018-04-04:
What I meant is, if your experiment does not need a core device, then you do not need a core device for the dashboard either. The experiment you are currently running needs a core device.

mingshenli, 2018-04-04:
so the dashboard can not simulate the board?

sbourdeauducq, 2018-04-04:
There is limited simulation support (see ``artiq/examples/no_hardware/repository/simple_simulation.py``), but no one finds it very useful, so it is not very developed and in particular not integrated with the dashboard.

sbourdeauducq, 2018-04-04:
Arguably, blinking the TTL panel when no device is connected and an experiment with the core device is run is not a useful feature.

mingshenli, 2018-04-04:
all right, I will get a kc705 and test it.(unluckily I have to wait for few days and that is why I hope the dashboard can simulate the board).

sbourdeauducq, 2018-04-04:
Try experiments that do not need the core device and play with the datasets and applets, see e.g. ``artiq/examples/no_hardware/repository/flopping_f_simulation.py``.

mingshenli, 2018-04-04:
that seems work well.

mingshenli, 2018-04-04:
by the way, where can I find the pin setting in the documentation so that I can know where to use a Oscilloscope to detect the ttl out? Did the documentation online say it?

sbourdeauducq, 2018-04-04:
https://github.com/m-labs/artiq/blob/master/artiq/gateware/nist_clock.py
https://github.com/m-labs/artiq/blob/master/artiq/gateware/nist_qc2.py
and the manual.

sbourdeauducq, 2018-04-04:
Note that ttl, 0 in nist_clock is not necessarily ttl0 in the device database (check the manual).

mingshenli, 2018-04-09:
I find that 'binaries' is in the wrong dictionary.

mingshenli, 2018-04-09:
it suppose to be C:\Users\18926\AppData\Local\conda\conda\envs\artiq-new\lib\site-packages\artiq\binaries,
but now in C:\Users\18926\AppData\Local\conda\conda\envs\artiq-new\lib\python3.5\site-packages\artiq

sbourdeauducq, 2018-04-09:
Can you give more details? What exactly did you do?

mingshenli, 2018-04-13:
well， I 'm not sure what happened, maybe there are too many version artiq on my computer so the artiq-new raise some error. However, now I remove all the old artiq and install the 3.6 and all the problem solved.

closed 2018-04-13 by mingshenli

mingshenli, 2018-04-13:
Yes I am in it. thanks a lot.