m-labs/artiq #978: TTL event replacement not working correctly

kesht123, 2018-04-03:
I am running the following experiment with gateware and software at commit [f077176](https://github.com/m-labs/artiq/commit/f0771765c17c53ded102d7d06ed09b609b316528), and looking at the SERDES TTL signals 422l and 422u on a scope. Our target is based off the kc705 NIST clock with some additional TTLs

```python
from artiq.experiment import *

class TestArtiqBug(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("_422u")
        self.setattr_device("_422l")

    @kernel
    def krun(self):
        self.core.reset()
        for _ in range(10000):
            delay(1*ms)
            self._422u.on()
            self._422l.on()
            delay(1*ms)
            self._422l.on()
            self._422l.off()
            self._422u.off()

    def run(self):
        self.krun()
```
The expected behaviour is that both the 422u and 422l TTLs should pulse together for 1 ms every 1 ms. Instead I see the 422u pulse correctly, but the 422l fail to switch off every other time in the loop, so that the pulses are now 3 ms wide. That is, I see that line 18 "self._422l.off()" only replaces line 17 "self._422l.on()" every other time in the loop. This problem occurs only when I schedule these two events (which act on the same TTL) at the same timestamp. The problem goes away when I introduce a delay between them.

The artiq_coreanalyzer trace shows the expected bug-free behaviour, which does not match reality (i.e. what is actually produced on the TTLs as measured by the scope). I believe the GUI does match reality.

jordens, 2018-04-03:
Thanks! Also for the clean repro. I think I have seen this as well in December with Kasli. So likely reproduced/ible (back then there were more pressing issues so I didn't isolate it).

cjbe, 2018-04-09:
@sbourdeauducq @jordens have either of you managed to reproduce this?

jordens, 2018-04-09:
Yes. I believe I have seen this in December. I haven't reproduced it with your code. @sbourdeauducq is looking at it.

sbourdeauducq, 2018-05-02:
Can be minimized further and reproduced in simulation:
```
diff --git a/artiq/gateware/test/rtio/test_sed_top.py b/artiq/gateware/test/rtio/test_sed_top.py
index d5de8897..09c740d8 100644
--- a/artiq/gateware/test/rtio/test_sed_top.py
+++ b/artiq/gateware/test/rtio/test_sed_top.py
@@ -73,7 +73,7 @@ def simulate(input_events):
 
     run_simulation(dut, {"sys": [
         gen(), monitor(),
-        (None for _ in range(45))
+        (None for _ in range(max(ts for ts, _ in input_events) + 15))
     ]}, {"sys": 5, "rio": 5, "rio_phy": 5})
 
     return ttl_changes, access_results
@@ -86,3 +86,17 @@ class TestSED(unittest.TestCase):
         ttl_changes, access_results = simulate(input_events)
         self.assertEqual(ttl_changes, [e[0] + latency for e in input_events])
         self.assertEqual(access_results, [("ok", 0)]*len(input_events))
+
+    def test_replace(self):
+        input_events = []
+        now = 19
+        for i in range(5):
+            now += 10
+            input_events += [(now, 1)]
+            now += 10
+            input_events += [(now, 1), (now, 0)]
+            #input_events += [(now, 0)]
+
+        ttl_changes, access_results = simulate(input_events)
+        self.assertEqual(access_results, [("ok", 0)]*len(input_events))
+        print(ttl_changes)
```

commit 83fb431c (2018-05-02): rtio/sed: pass sequence numbers through the FIFOs. Closes #978
closed 2018-05-02 by sbourdeauducq