m-labs/artiq #979: Runtime panic after device has been up for a long time

cjbe, 2018-04-04:
I occasionally see runtime panics like the below on Kasli. These only seem to occur after the device has been up for a while (23 hours in this case). At the same time I see some sort of slow-down of the core device - normally the experiment below takes 0.8s to run, just before it crashed it was taking ~5s to run, the majority of which occurred after the "Mismatch between gateware and software" warning. I did not manage to get a packet dump before it crashed - I will try next time.

This does not seem to be specific to the experiment I am running - I have seen this happen with several different experiments of different character.

```
[ 83908.784314s]  INFO(runtime::session): no connection, starting idle kernel
[ 83911.994893s]  INFO(runtime::session): new connection from 10.255.6.191:53172
[ 83912.237798s]  INFO(runtime::session): idle kernel interrupted
[ 83914.722352s]  INFO(runtime::session): no connection, starting idle kernel
panic at src/libcore/result.rs:906: already borrowed: BorrowMutError
backtrace for software version 4.0.dev+1274.g8182f97a:
0x40028260
0x40049db0
0x4002b344
0x4002b9b8
0x4001bce0
0x40023fbc
0x40027fe0
halting.
```

```python
class SetDds(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.dds = self.get_device("_1092")
        self.att = self.get_argument("Attenuator",
                    NumberValue(29.0, max=29., min=0., unit="dB", scale=1.))
        self.freq = self.get_argument("Frequency",
                    NumberValue(150*MHz, min=10*MHz, max=300*MHz, unit="MHz"))
        self.amp = self.get_argument("Amplitude", 
                    NumberValue(1.0, min=0., max=1.))
        self.sw = self.get_argument("Switch", BooleanValue(False))

    @kernel
    def run(self):
        self.core.break_realtime()
        delay(1*ms)

        self.dds.set_att(self.att)
        self.dds.set(self.freq, amplitude=self.amp)

        if self.sw:
            self.dds.sw.on()
        else:
            self.dds.sw.off()
```

sbourdeauducq, 2018-04-04:
Can you run this through addr2line?
i.e. ``or1k-linux-addr2line -ipfC -e runtime.elf 0x40028260 0x40049db0 ...``

cjbe, 2018-04-04:
```
$ or1k-linux-addr2line -ipfC -e runtime.elf 0x40028260 0x40049db0 0x4002b344 0x4002b9b8 0x4001bce0 0x40023fbc 0x40027fe0
board::config::imp::read<closure,bool> at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/libboard/config.rs:122
 (inlined by) board::config::imp::read_str<closure,bool> at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/libboard/config.rs:137
 (inlined by) runtime::panic_fmt at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/main.rs:294
core::panicking::panic_fmt at /var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/panicking.rs:71
core::result::unwrap_failed<core::cell::BorrowMutError> at /var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/macros.rs:23
core::result::{{impl}}::expect<core::cell::RefMut<log_buffer::LogBuffer<&mut [u8]>>,core::cell::BorrowMutError> at /var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/result.rs:799
 (inlined by) core::cell::{{impl}}::borrow_mut<log_buffer::LogBuffer<&mut [u8]>> at /var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/cell.rs:776
 (inlined by) logger_artiq::{{impl}}::is_empty at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/liblogger_artiq/lib.rs:48
runtime::mgmt::worker::{{closure}}::{{closure}} at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/mgmt.rs:44
 (inlined by) logger_artiq::{{impl}}::with<bool,closure> at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/liblogger_artiq/lib.rs:40
 (inlined by) runtime::mgmt::worker::{{closure}} at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/mgmt.rs:44
runtime::sched::{{impl}}::run at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/sched.rs:146
 (inlined by) runtime::startup_ethernet at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/main.rs:233
runtime::startup at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/main.rs:107
 (inlined by) core::ops::function::FnOnce::call_once<fn(),()> at /var/lib/buildbot/slaves/conda-lin64/miniconda/conda-bld/rust-core-or1k_1515447451680/work/src/libcore/ops/function.rs:223
 (inlined by) logger_artiq::{{impl}}::register<fn()> at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/liblogger_artiq/lib.rs:36
 (inlined by) runtime::main at /home/bb/worker/kasli_hoa2_master/build/artiq/firmware/runtime/main.rs:266
```

sbourdeauducq, 2018-04-04:
Do you have ``aqctl_corelog`` running?

cjbe, 2018-04-04:
Yes, `aqctl_corelog` is running. 

I will stop it running now and see if I still see this bug, however it will take a while to get statistics - this problem has only occured 2 times in the last 7 days.

jordens, 2018-04-04:
this is https://github.com/cjbe/artiq/commit/8182f97a1238be94d67feb0f029e8d566516b251

whitequark, 2018-04-05:
> This may be a bug in the logging server
> At the same time I see some sort of slow-down of the core device - normally the experiment below takes 0.8s to run, just before it crashed it was taking ~5s to run, the majority of which occurred after the "Mismatch between gateware and software" warning.

You're correct and there is an easily addressable bug in the logger server; it's a race condition and exposed by the slowdown. However, the slowdown issue is significantly more important, and likely harder to address. The only cause for such slowdown in a long running core device I can find is heap fragmentation, and we don't currently have the tools to either diagnose or fix it.

I'll add a diagnostic option to print heap contents, but so far I don't have any good ideas for actually fixing fragmentation with reasonable effort.

jordens, 2018-04-07:
Reproduced easily on master. This is bad.

jordens, 2018-04-07:
Err. Wrong issue.

whitequark, 2018-04-07:
@jordens Ack. What are the exact steps for reproduction you're taking?

jordens, 2018-04-07:
@whitequark were you unable to reproduce it with the code from @cjbe or did you not try yet?
I my case it's an experiment that does async rpcs at ~20 Hz run under `artiq_master` with the dashboard doing the usual moninj traffic. Kasli/opticlock.

whitequark, 2018-04-07:
@jordens I did not try yet. How long does it need to run?

jordens, 2018-04-10:
@whitequark Please give it a try. I have seen the slowdown after a couple of hours. If it is heap fragmentation you should be able to accelerate the degradation. Let me know if you are unable to triage or reproduce it.

whitequark, 2018-04-20:
The panics won't happen anymore. Investigating the slowdown.

whitequark, 2018-04-21:
@jordens I am unable to reproduce this on master. Here is the experiment I am using, running it from the dashboard:
```python
import time

from artiq.experiment import *

class AsyncRPCStress(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.time = time.clock()
        self.count = 0

    @rpc(flags={"async"})
    def sink(self, data):
        self.count += 1
        now = time.clock()
        if now - self.time > 5.0:
            print("{:.1f} rpc/s".format(self.count / (now - self.time)))
            self.time = now
            self.count = 0

    @kernel
    def run(self):
        while True:
            self.sink(0)
```

I consistently get around 24000 rpc/s. Also, after an initial settle down period, there is no change in heap layout at all (zero line diff) with ~100k RPCs in between, unless you catch it during some transient allocation. You can ask the core device for the current heap layout using `artiq_coredebug allocator`, and it will print it to the UART (not the log).

What is the exact code you were using?

jordens, 2018-04-22:
Are you trying this with artiq_run or artiq_master?

whitequark, 2018-04-22:
artiq_master, like you said.

jordens, 2018-04-22:
And with the dashboard running as well?

jordens, 2018-04-22:
@whitequark Ack. I'll see whether I can isolate it.

whitequark, 2018-05-05:
@jordens I've fixed #986. In case you manage to reproduce this, you can gather and view a profile with:
```
artiq_coreprofile start
artiq_run [...]
artiq_coreprofile save artiq.prof /tmp/artiq_kc705/nist_clock/software/runtime/runtime.elf # e.g.
kcachegrind artiq.prof
```

whitequark, 2018-05-16:
@cjbe I am not able to reproduce this issue. Can you please try again on master? And in case it does happen, can you capture a profile as described in https://github.com/m-labs/artiq/issues/979#issuecomment-386769720?

cjbe, 2018-05-30:
@whitequark I have not managed to reproduce this since the fix for #986 - if I do see this again, I will capture a profile.