m-labs/artiq #980: Dashboard causes 'asyncio:Fatal write error on socket transport' occasionally

cjbe, 2018-04-06:
When running the dashboard, I occasionally see 'asyncio:Fatal write error on socket transport' errors. I have not found a way to reliably reproduce this.

I am running from the master branch (dd21c07b85c1067913d521b09fda80acf561df7d), with the master on Linux, and the dashboard on Windows 7. These errors have been occurring for quite a while - they have not been recently introduced. 

Full message:
```
asyncio:Fatal write error on socket transport
protocol: <asyncio.streams.StreamReaderProtocol object at 0x7fcdfc087cf8>
transport: <_SelectorSocketTransport fd=44 read=idle write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/home/ion/miniconda3/envs/artiq/lib/python3.5/asyncio/selector_events.py", line 761, in write
    n = self._sock.send(data)
TimeoutError: [Errno 110] Connection timed out
```

jordens, 2018-04-07:
Is this #979 or is it between the dashboard and the master?
And it is also not just an old connection to a nonexistent dashboard istance or applet timing out?

cjbe, 2018-04-07:
I believe this is between the dashboard and master, as I only see these errors when a dashboard is running. I see these errors without any applets started.  I don't know if this is due to old connections to a nonexistent dashboard instance.

sbourdeauducq, 2018-04-07:
What prints this, the master?

sbourdeauducq, 2018-04-07:
Try enabling the debug mode of asyncio, which may or may not give more information.
https://docs.python.org/3/library/asyncio-dev.html

cjbe, 2018-04-07:
@sbourdeauducq the master prints this. 
I've enabled debug and will see what I get out.