m-labs/artiq #984: Conda problem about install artiq 4.0.dev

KaifengC, 2018-04-19:
I'm using a window PC and trying to install artiq 4.0.dev from the `dev` channel but seems something goes wrong:
[image]

Then I found I only got an `artiq\binaries\kc705-nist_clock` folder installed in this new environment.
I tried to install other versions and found out that this problem first appeared in artiq 3.5.

I guess there's something wrong about the conda setting?

whitequark, 2018-04-19:
This should have been fixed in 1ef673c2d43c3, but conda is selecting the commit just before that for some reason.

sbourdeauducq, 2018-05-13:
@KaifengC Do you still have this problem?

KaifengC, 2018-05-14:
It's still the same. But using `conda install artiq=4.0.dev` in the activated environment makes it work.

whitequark, 2018-05-25:
Can you `conda update conda` and try again?

KaifengC, 2018-05-25:
I use `conda update  conda` and update conda to 4.5.4-py36_0. Then I create a new environment, still the same.

```bash
D:\>conda update conda
Fetching package metadata .................
Solving package specifications: .

Package plan for installation in environment D:\ProgramData\Anaconda3:

The following NEW packages will be INSTALLED:

    ca-certificates: 2018.4.16-0   conda-forge/label/main
    vc:              14-0          conda-forge/label/main

The following packages will be UPDATED:

    conda:           4.3.14-py36_1                        --> 4.5.4-py36_0  con
a-forge/label/main
    openssl:         1.0.2k-vc14_0                        --> 1.0.2o-vc14_0 con
a-forge/label/main [vc14]
    pycosat:         0.6.1-py36_1                         --> 0.6.3-py36_0  con
a-forge/label/main

Proceed ([y]/n)? y

pycosat-0.6.3- 100% |###############################| Time: 0:00:01  60.07 kB/s
conda-4.5.4-py 100% |###############################| Time: 0:00:12  55.41 kB/s

D:\>conda create -n artiq_4-2 artiq-kc705-nist_clock=4.0.dev
Solving environment: done

## Package Plan ##

  environment location: D:\ProgramData\Anaconda3\envs\artiq_4-2

  added / updated specs:
    - artiq-kc705-nist_clock=4.0.dev

The following NEW packages will be INSTALLED:

    artiq-kc705-nist_clock: 4.0.dev-820+gitbb90fb7d m-labs/label/dev

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use:
# > activate artiq_4-2
#
# To deactivate an active environment, use:
# > deactivate
#
# * for power-users using bash, you must source
#

D:\>activate artiq_4-2

(artiq_4-2) D:\>conda list
# packages in environment at D:\ProgramData\Anaconda3\envs\artiq_4-2:
#
# Name                    Version                   Build  Channel
artiq-kc705-nist_clock    4.0.dev         820+gitbb90fb7d    m-labs/label/dev
```

sbourdeauducq, 2018-06-04:
This is also affecting 3.6.

sbourdeauducq, 2018-08-18:
Fixed by updating conda-build + https://github.com/m-labs/artiq/commit/d6992f6a0c83f1930197d0ab6fbb3b538bb4d231

closed 2018-08-18 by sbourdeauducq