m-labs/artiq #992: SU-Servo: ADC/IIR max/min values

hartytp, 2018-05-01:
We need a means of checking whether any servo channel/profile has railed, and also of logging the IIR input/outputs over time (e.g. to watch for head room slowly decreasing due to fibre coupling efficiency degrading over time or whatever).

The way I'd imagine doing this (comments welcome) is:
- for each channel/profile, we store the max/min IIR input (ADC value) and output (DDS ASD). These should clear on read, so that each read gives the max/min value since the last read. These max/min values should only update when the servo is running (integrator active). I think we previously discussed having just a "railed" indicator, but on reflection, max/min values would be nicer. 
- one schedules an experiment to run every 10s of seconds to minutes, which reads the max/min value for each profile one cares about and broadcasts them.
- a GUI/log can then pick these up and do something with them

jordens, 2018-05-02:
That implementation has a couple issues. It would require a lot of restructuring, it doesn't fit the memory layout, the pipelining would suffer, the czlycle length would increase. It would need 32*8*2+2*8 words storage. And it's really tricky to get the clearing and readout working given the collisions. And there are a couple conceptual issues as well. PI ringing can give you transient clips that would be false positive. Usually you'd want to know whether it rails at certain times. With the latching min max you'd have to interact twice with the phy to clear and then later to read them.

Let's think about what's already there and what we can do easily to enable your use case.

Right now you can read out the ADC and ASF (iir x and y) at any time and decide whether it is good or bad and decide about changing gains and attenuators. You can log that and build a gui on top of it with the available tooling in artiq.

I can add a clear-on-read railed indicator to the output channels (but not per profile). That should allow you to detect transient clips if you need that and to to detect any other problems caused by railing or drifting things.

hartytp, 2018-05-02:
@jordens thanks for the feedback. Points taken.

> I can add a clear-on-read railed indicator to the output channels (but not per profile). That should allow you to detect transient clips if you need that and to to detect any other problems caused by railing or drifting things.

That works for me. @cjbe?

cjbe, 2018-06-01:
@jordens sounds good to me.

closed 2018-06-01 by jordens