m-labs/artiq #996: serwb breaks when AMC bitstream is built with SAWG

sbourdeauducq, 2018-05-07:
```
[    40.229201s]  INFO(board_artiq::serwb): done.
[    40.232251s]  INFO(board_artiq::serwb): RTM to AMC Link test
[    41.237899s]  INFO(board_artiq::serwb): 209716 errors
[    41.241665s]  INFO(board_artiq::serwb): AMC to RTM Link test
```
[freezes]

Same behavior with two different AMCs.

enjoy-digital, 2018-05-07:
https://github.com/m-labs/artiq/commit/60fd362d57348fe424410e7e95323fac867ce0e5 was the only thing i was able to found. (Error while refactoring the code to simplify clocking).

I was not able to test on hardware since M-Labs's RTM is no longer working.

sbourdeauducq, 2018-05-08:
What do you mean by no longer working?

enjoy-digital, 2018-05-08:
OK then i'll retry on Wednesday, i was using 3:10, it was working correctly but then stopped.

sbourdeauducq, 2018-05-08:
What was the error?

sbourdeauducq, 2018-05-08:
Anyway I tested the latest code, and seems this problem is resolved. Thanks.

closed 2018-05-08 by sbourdeauducq