m-labs/artiq #997: serwb does not meet timing

sbourdeauducq, 2018-05-10:
```
Max Delay Paths
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -0.044ns  (required time - arrival time)
  Source:                 storage_1_reg_0/CLKARDCLK
                            (rising edge-triggered cell RAMB36E1 clocked by sys_clk  {rise@0.000ns fall@4.000ns period=8.000ns})
  Destination:            storage_2_reg_0_15_6_11/RAMC_D1/I
                            (rising edge-triggered cell RAMD32 clocked by sys_clk  {rise@0.000ns fall@4.000ns period=8.000ns})
  Path Group:             sys_clk
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            8.000ns  (sys_clk rise@8.000ns - sys_clk rise@0.000ns)
  Data Path Delay:        7.667ns  (logic 4.441ns (57.923%)  route 3.226ns (42.077%))
  Logic Levels:           6  (CARRY4=3 LUT4=1 LUT6=2)
  Clock Path Skew:        -0.068ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    7.129ns = ( 15.129 - 8.000 ) 
    Source Clock Delay      (SCD):    7.714ns
    Clock Pessimism Removal (CPR):    0.517ns
  Clock Uncertainty:      0.059ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.095ns
    Phase Error              (PE):    0.000ns

    Location             Delay type                Incr(ns)  Path(ns)    Netlist Resource(s)
  -------------------------------------------------------------------    -------------------
                         (clock sys_clk rise edge)    0.000     0.000 r  
    R18                                               0.000     0.000 r  amc_rtm_serwb_clk_p (IN)
                         net (fo=0)                   0.000     0.000    amc_rtm_serwb_clk_p
    R18                  IBUFDS (Prop_ibufds_I_O)     0.942     0.942 r  IBUFDS/O
                         net (fo=1, routed)           0.900     1.842    serwb_phy_rtm_serdes_clk_i
    BUFR_X0Y1            BUFR (Prop_bufr_I_O)         0.512     2.354 r  BUFR_1/O
                         net (fo=1, routed)           0.754     3.108    serwb_phy_rtm_serdes_clk_i_bufr
    BUFGCTRL_X0Y1        BUFG (Prop_bufg_I_O)         0.096     3.204 r  BUFG_1/O
                         net (fo=1, routed)           1.574     4.779    serwb_refclk
    MMCME2_ADV_X0Y0      MMCME2_ADV (Prop_mmcme2_adv_CLKIN1_CLKOUT0)
                                                      0.088     4.867 r  MMCME2_BASE/CLKOUT0
                         net (fo=2, routed)           1.025     5.892    pll_sys4x
    BUFR_X0Y0            BUFR (Prop_bufr_I_O)         0.982     6.874 r  BUFR/O
                         net (fo=1275, routed)        0.840     7.714    sys_clk
    RAMB36_X0Y2          RAMB36E1                                     r  storage_1_reg_0/CLKARDCLK
  -------------------------------------------------------------------    -------------------
    RAMB36_X0Y2          RAMB36E1 (Prop_ramb36e1_CLKARDCLK_DOADO[5])
                                                      2.454    10.168 r  storage_1_reg_0/DOADO[5]
                         net (fo=6, routed)           1.232    11.401    serwb_core_record_sender_pbuffer_fifo_out_payload_count[4]
    SLICE_X10Y12         LUT6 (Prop_lut6_I0_O)        0.124    11.525 r  storage_2_reg_0_15_0_5_i_36/O
                         net (fo=4, routed)           0.466    11.990    storage_2_reg_0_15_0_5_i_36_n_0
    SLICE_X8Y12          LUT4 (Prop_lut4_I2_O)        0.124    12.114 r  storage_2_reg_0_15_0_5_i_31/O
                         net (fo=1, routed)           0.000    12.114    storage_2_reg_0_15_0_5_i_31_n_0
    SLICE_X8Y12          CARRY4 (Prop_carry4_S[3]_CO[3])
                       net (fo=1, routed)           0.000    12.490    storage_2_reg_0_15_0_5_i_21_n_0
    SLICE_X8Y13          CARRY4 (Prop_carry4_CI_O[0])
                                                      0.219    12.709 r  storage_2_reg_0_15_6_11_i_17/O[0]
                         net (fo=1, routed)           0.647    13.356    storage_2_reg_0_15_6_11_i_17_n_7
    SLICE_X9Y13          CARRY4 (Prop_carry4_S[0]_O[2])
                                                      0.842    14.198 r  storage_2_reg_0_15_6_11_i_12/O[2]
                         net (fo=1, routed)           0.411    14.609    storage_2_reg_0_15_6_11_i_12_n_5
    SLICE_X9Y14          LUT6 (Prop_lut6_I2_O)        0.302    14.911 r  storage_2_reg_0_15_6_11_i_5/O
                         net (fo=1, routed)           0.471    15.381    storage_2_reg_0_15_6_11/DIC1
    SLICE_X8Y15          RAMD32                                       r  storage_2_reg_0_15_6_11/RAMC_D1/I
  -------------------------------------------------------------------    -------------------

                         (clock sys_clk rise edge)    8.000     8.000 r  
    R18                                               0.000     8.000 r  amc_rtm_serwb_clk_p (IN)
                         net (fo=0)                   0.000     8.000    amc_rtm_serwb_clk_p
    R18                  IBUFDS (Prop_ibufds_I_O)     0.899     8.899 r  IBUFDS/O
                         net (fo=1, routed)           0.763     9.662    serwb_phy_rtm_serdes_clk_i
    BUFR_X0Y1            BUFR (Prop_bufr_I_O)         0.487    10.149 r  BUFR_1/O
                         net (fo=1, routed)           0.713    10.862    serwb_phy_rtm_serdes_clk_i_bufr
    BUFGCTRL_X0Y1        BUFG (Prop_bufg_I_O)         0.091    10.953 r  BUFG_1/O
                         net (fo=1, routed)           1.455    12.408    serwb_refclk
    MMCME2_ADV_X0Y0      MMCME2_ADV (Prop_mmcme2_adv_CLKIN1_CLKOUT0)
                                                      0.083    12.491 r  MMCME2_BASE/CLKOUT0
                         net (fo=2, routed)           0.960    13.451    pll_sys4x
    BUFR_X0Y0            BUFR (Prop_bufr_I_O)         0.918    14.369 r  BUFR/O
                         net (fo=1275, routed)        0.760    15.129    storage_2_reg_0_15_6_11/WCLK
    SLICE_X8Y15          RAMD32                                       r  storage_2_reg_0_15_6_11/RAMC_D1/CLK
                         clock pessimism              0.517    15.646    
                         clock uncertainty           -0.059    15.587    
    SLICE_X8Y15          RAMD32 (Setup_ramd32_CLK_I)
                                                     -0.249    15.338    storage_2_reg_0_15_6_11/RAMC_D1
  -------------------------------------------------------------------
                         required time                         15.338    
                         arrival time                         -15.381    
  -------------------------------------------------------------------
                         slack                                 -0.044    
```

hartytp, 2018-05-10:
@sbourdeauducq is it worth putting a bigger/faster FPGA on the RTM for Sayma v2.0? Obviously, it's best to use the cheapest FPGA we can, but I wonder if a bit more money on FPGAs will save us a lot of development time in the future?

sbourdeauducq, 2018-05-10:
This is, above all, about keeping high-quality FPGA code in ARTIQ, not shaving a few dollars from the board.

enjoy-digital, 2018-05-11:
This timing error appeared with new version of Vivado. I haven't done any specifc optimization on the RTM side before since Vivado < 2018.1 was not complaining. I'm looking at that.

commit 2e3bf860 (2018-05-11): serwb: reduce buffering. Closes #997
closed 2018-05-11 by sbourdeauducq