m-labs/artiq #998: RTIO sequence error when resetting two SAWG channels

sbourdeauducq, 2018-05-10:
Kernel:
```
from artiq.experiment import *

class SAWGResetSequenceError(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        self.sawg0.reset()
        self.sawg1.reset()
```

When run on Sayma this produces the following error:
```
[   757.940973s]  INFO(runtime::kern_hwreq): resetting RTIO
[   757.945261s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 22
```

sbourdeauducq, 2018-05-10:
The problem can be worked around by inserting a delay (e.g. 80ns) between the two resets.

jbqubit, 2018-05-10:
Does it make sense for xxx.reset() to automatically include whatever delays are necessary? AFAICT resets are done at the start of experiment cycles where the simplicity of just-works (without fiddling with delays) is nice.

sbourdeauducq, 2018-05-17:
What I don't understand is why there is no sequence error with a single ``reset()``. The end of ``reset()`` calls ``set_mu`` 9 times without advancing the time line, which is more than the number of SED lanes, so there should be a sequence error already.
With TTLs, sequence errors behave as expected and can be tested with:
```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.ttls = [self.get_device("ttl"+str(i)) for i in range(8)]
        self.ttls.append(self.get_device("ttl_sma"))
        self.ttls.append(self.get_device("led"))

    @kernel
    def run(self):
        self.core.reset()
        delay(5*ms)
        for ttl in self.ttls:
            ttl.on()
```

sbourdeauducq, 2018-05-25:
@jordens Do you have a repro on Kasli?

jordens, 2018-05-25:
Not right now.

jbqubit, 2018-05-30:
OK. Confirmed that delays work.

sbourdeauducq, 2018-05-31:
And do it again with just ``sawg0`` being initialized (comment the ``sawg1`` reset).

jbqubit, 2018-05-31:
Please make mods to your hardware to get Ethernet to work. Please open an Issue on sinara-hw if you don't know what changes need to be made.

sbourdeauducq, 2018-05-31:
I know what changes need to be made, but they take hours. Whereas it should take you a few minutes to get the dump.

sbourdeauducq, 2018-05-31:
And I don't even have a SFP-SATA adapter anymore as I sent it to you.

jbqubit, 2018-05-31:
These things work in uTCA crate, right? @gkasprow 

How do I clear the analyzer buffer? It's not mentioned here.

https://m-labs.hk/artiq/manual-master/utilities.html?highlight=analyzer#core-device-rtio-analyzer-tool

hartytp, 2018-05-31:
> Please make mods to your hardware to get Ethernet to work. Please open an Issue on sinara-hw if you don't know what changes need to be made.

Joe, is it really worth M-Labs doing this? Greg's verified that ethernet works reliably with these changes on several boards. For the time being, I'd argue that doing this rework would not be a good use of @sbourdeauducq's time.

sbourdeauducq, 2018-05-31:
> These things work in uTCA crate, right?

No they don't.

> For the time being, I'd argue that doing this rework would not be a good use of @sbourdeauducq's time.

Exactly, also I hope we can get a repro on Kasli after this, where it is much easier to track down problems.

jbqubit, 2018-05-31:
@hartytp M-Labs needs a working hardware setup so they have the ability to debug Sayma. Many of the very nice debug tools rely on Ethernet. It's annoying the M-Labs sent their only fully functioning board to Duke without having a fully-functioning backup in HK. It's like the warning on aircraft to "put the oxygen mask on yourself before assisting your fellow passenger."

sbourdeauducq, 2018-05-31:
And the analyzer buffer is cleared every time you use ``artiq_coreanalyzer`` to retrieve it from the board. So the procedure is: artiq_coreanalyzer, throw away the results, run experiment, artiq_coreanalyzer again and post results. Or if the board is freshly booted and without a startup kernel, the buffer is already empty.

jbqubit, 2018-05-31:
```
from artiq.experiment import *

class SAWGResetSequenceError(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        self.sawg0.reset()
        self.sawg1.reset()
```

```
$ history
 2557  artiq_coreanalyzer 
 2558  artiq_run repository/test_sawg_reset.py 
 2559  artiq_coreanalyzer -d dump.raw
 2560  zip dump.raw.zip dump.raw
```

```
[  1169.185338s]  INFO(runtime::session): new connection from 192.168.1.68:55492
[  1169.222844s]  INFO(runtime::kern_hwreq): resetting RTIO
[  1169.227171s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 22
[  1169.236665s]  INFO(runtime::session): no connection, starting idle kernel
[  1169.242381s]  INFO(runtime::session): no idle kernel found
```

[dump.raw.zip](https://github.com/m-labs/artiq/files/2058731/dump.raw.zip)

jbqubit, 2018-05-31:
> And the analyzer buffer is cleared every time you use artiq_coreanalyzer to retrieve it from the board. So the procedure is: artiq_coreanalyzer, throw away the results, run experiment, [...]

This would be really helpful to have in ARTIQ documentation. :)

sbourdeauducq, 2018-05-31:
Thanks. Can you do another one with ``sawg1.reset`` commented out?

jbqubit, 2018-05-31:
```
from artiq.experiment import *

class SAWGResetSequenceError(EnvExperiment):
    def build(self):
        self.setattr_device("core")

        self.setattr_device("sawg0")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        self.sawg0.reset()
        # self.sawg1.reset()
```

```
 2578  artiq_coreanalyzer
 2579  artiq_run repository/test_sawg_reset.py 
 2580  artiq_coreanalyzer -d dump.raw
 2581  zip dump.raw.zip dump.raw
```

```
[  1503.298083s]  INFO(runtime::session): new connection from 192.168.1.68:55604
[  1503.540631s]  INFO(runtime::kern_hwreq): resetting RTIO
[  1503.547473s]  INFO(runtime::session): no connection, starting idle kernel
[  1503.553408s]  INFO(runtime::session): no idle kernel found
[  1521.175271s]  INFO(runtime::analyzer): connection from 192.168.1.68:42686
```

[dump.raw.zip](https://github.com/m-labs/artiq/files/2058762/dump.raw.zip)

jbqubit, 2018-05-31:
Also, you're welcome to run my Sayma remotely while we're all asleep in the USA. :)

sbourdeauducq, 2018-05-31:
So, there are two things that are wrong with those dumps:
1. the first one (sawg0+sawg1) contains tons of RTIO events that should not be there.
2. the second one (sawg0) should produce an underflow.

I'll replay 2) on Kasli and see what happens.

sbourdeauducq, 2018-05-31:
> contains tons of RTIO events that should not be there.

There are 18 seconds between the submission of the first and last events in that dump. Looks like that can be explained by the buffer not being cleared, due to mistake or ARTIQ bug.

jbqubit, 2018-05-31:
You saw history of what I typed. Now you have remote access so can run yourself. :)

sbourdeauducq, 2018-06-01:
With the same RTIO sequence that corresponds to one SAWG reset (channels and timestamps shifted), Kasli behaves itself and correctly reports the underflow.

```python
from collections import namedtuple

OutputMessage = namedtuple(
    "OutputMessage", "channel timestamp rtio_counter address data")

evts = [
OutputMessage(channel=6, timestamp=224617565613, rtio_counter=224617291119, address=0, data=0),
OutputMessage(channel=6, timestamp=224617565616, rtio_counter=224617291372, address=1, data=7),
OutputMessage(channel=6, timestamp=224617565619, rtio_counter=224617291504, address=2, data=1),
OutputMessage(channel=6, timestamp=224617565622, rtio_counter=224617299780, address=6, data=4294947398),
OutputMessage(channel=6, timestamp=224617565625, rtio_counter=224617303805, address=7, data=19898),
OutputMessage(channel=6, timestamp=224617565628, rtio_counter=224617307578, address=4, data=4294934529),
OutputMessage(channel=6, timestamp=224617565631, rtio_counter=224617311276, address=5, data=32767),
OutputMessage(channel=14, timestamp=224617565634, rtio_counter=224617311472, address=0, data=0),
OutputMessage(channel=9, timestamp=224617565634, rtio_counter=224617311582, address=0, data=0),
OutputMessage(channel=12, timestamp=224617565634, rtio_counter=224617311698, address=0, data=0),
OutputMessage(channel=15, timestamp=224617565634, rtio_counter=224617311796, address=0, data=0),
OutputMessage(channel=10, timestamp=224617565634, rtio_counter=224617311902, address=0, data=0),
OutputMessage(channel=13, timestamp=224617565634, rtio_counter=224617311990, address=0, data=0),
OutputMessage(channel=8, timestamp=224617565634, rtio_counter=224617312106, address=0, data=0),
OutputMessage(channel=11, timestamp=224617565634, rtio_counter=224617312196, address=0, data=0),
OutputMessage(channel=7, timestamp=224617565634, rtio_counter=224617312312, address=0, data=0),
]

for evt in evts:
	print("rtio_output(st+{time_mu}, {channel}, {addr}, {data})".format(
		time_mu=(evt.timestamp-224617565613)*3,
		channel=evt.channel-6,
		addr=evt.address,
		data=evt.data))
```

```python
from artiq.experiment import *
from artiq.coredevice.rtio import rtio_output

class RTIOBug(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
        st = now_mu()
        rtio_output(st+0, 0, 0, 0)
        rtio_output(st+24, 0, 1, 7)
        rtio_output(st+48, 0, 2, 1)
        rtio_output(st+72, 0, 6, 4294947398)
        rtio_output(st+96, 0, 7, 19898)
        rtio_output(st+120, 0, 4, 4294934529)
        rtio_output(st+144, 0, 5, 32767)
        rtio_output(st+168, 8, 0, 0)
        rtio_output(st+168, 3, 0, 0)
        rtio_output(st+168, 6, 0, 0)
        rtio_output(st+168, 9, 0, 0)
        rtio_output(st+168, 4, 0, 0)
        rtio_output(st+168, 7, 0, 0)
        rtio_output(st+168, 2, 0, 0)
        rtio_output(st+168, 5, 0, 0)
        rtio_output(st+168, 1, 0, 0)
```

```
[ 74910.985828s]  INFO(runtime::session): new connection from 192.168.1.100:50700
[ 74911.022571s]  INFO(runtime::kern_hwreq): resetting RTIO
[ 74911.026761s] ERROR(runtime::rtio_mgt): RTIO sequence error involving channel 1
```

sbourdeauducq, 2018-06-01:
Could be a simple bug in SED with the handling of coarse/fine timestamps.

sbourdeauducq, 2018-06-01:
Confirmed that Sayma does **not** report the sequence error, both with the same code and without the *8 timestamp.

sbourdeauducq, 2018-06-01:
This produces the expected sequence error:
```python
from artiq.experiment import *
from artiq.coredevice.rtio import rtio_output

class RTIOBug(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        self.core.reset()
        delay(10*ms)
        st = now_mu()
        for i in range(9):
            rtio_output(st, 8, 0, 0)
```

Changing RTIO channel numbers by using ``rtio_output(st, 8+i, 0, 0)`` makes the sequence error disappear.

Here is the code that detects sequence errors. Where is the dependency on RTIO channels?!?! ``quash`` is only asserted for channel 86 (as can be verified in the generated Verilog).
https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/sed/lane_distributor.py#L130-L135

Kasli and KC705 have none of this insanity and correctly report the sequence error (respectively on channels 8 and 16) in both cases.

hartytp, 2018-06-03:
I do feel like there is something we're missing as there are a few odd bugs (crashes, dac glitches, etc) which don't make sense and don't seem very reproducible. Question is, if there is some common cause, how do we find it? More microscope probes?

sbourdeauducq, 2018-06-03:
Bug is still present with the JESD core removed. Oddly, this makes Vivado fail timing in several places.

hartytp, 2018-06-03:
Does removing the LOCs help? Can we remove those with appropriate timing constraints?

sbourdeauducq, 2018-06-04:
I don't think this has anything to do with the LOCs.

hartytp, 2018-06-10:
@sbourdeauducq do you need any other data to help debug this?

sbourdeauducq, 2018-06-23:
The Sayma DRTIO master, after adding SAWG/JESD, now also exhibits this bug.
Is the correlation with #1065 only a coincidence?

sbourdeauducq, 2018-06-23:
Reverting commit 83428961ad3dd74c5aa58859da035630c2fc06cf makes the bug disappear on the master (and stops the crash-kernel from crashing). So, it really looks like this and #1065 are linked.

sbourdeauducq, 2018-08-03:
Removing quash channels as below also does not prevent the bug.
```
diff --git a/artiq/gateware/rtio/sed/lane_distributor.py b/artiq/gateware/rtio/sed/lane_distributor.py
index 4f8b7e146..698954885 100644
--- a/artiq/gateware/rtio/sed/lane_distributor.py
+++ b/artiq/gateware/rtio/sed/lane_distributor.py
@@ -84,14 +84,6 @@ class LaneDistributor(Module):
             last_minus_timestamp.eq(last_coarse_timestamp - coarse_timestamp)
         ]
 
-        # Quash channels are "dummy" channels to which writes are completely ignored.
-        # This is used by the RTIO log channel, which is taken into account
-        # by the analyzer but does not enter the lanes.
-        quash = Signal()
-        self.sync += quash.eq(0)
-        for channel in quash_channels:
-            self.sync += If(self.cri.chan_sel[:16] == channel, quash.eq(1))
-
         assert all(abs(c) < 1 << 14 - 1 for c in compensation)
         latency_compensation = Memory(14, len(compensation), init=compensation)
         latency_compensation_port = latency_compensation.get_port()
@@ -127,7 +119,7 @@ class LaneDistributor(Module):
             ),
 
             timestamp_above_lane_min.eq(Mux(use_laneB, timestamp_above_laneB_min, timestamp_above_laneA_min)),
-            If(~quash & (self.cri.cmd == cri.commands["write"]),
+            If(self.cri.cmd == cri.commands["write"],
                 If(timestamp_above_min,
                     If(timestamp_above_lane_min,
                         do_write.eq(1)
```

sbourdeauducq, 2018-08-03:
There is no bug - I forgot that SAWG uses latency compensation extensively.

The actual timestamps during one reset can be obtained with:
```
from collections import namedtuple

compensation = [0, 0, 0, 0, 0, 0, 0, 50, 2, 0, 0, 2, 0, 0, 30, 30]

OutputMessage = namedtuple(
    "OutputMessage", "channel timestamp rtio_counter address data")

evts = [
OutputMessage(channel=6, timestamp=224617565613, rtio_counter=224617291119, address=0, data=0),
OutputMessage(channel=6, timestamp=224617565616, rtio_counter=224617291372, address=1, data=7),
OutputMessage(channel=6, timestamp=224617565619, rtio_counter=224617291504, address=2, data=1),
OutputMessage(channel=6, timestamp=224617565622, rtio_counter=224617299780, address=6, data=4294947398),
OutputMessage(channel=6, timestamp=224617565625, rtio_counter=224617303805, address=7, data=19898),
OutputMessage(channel=6, timestamp=224617565628, rtio_counter=224617307578, address=4, data=4294934529),
OutputMessage(channel=6, timestamp=224617565631, rtio_counter=224617311276, address=5, data=32767),
OutputMessage(channel=14, timestamp=224617565634, rtio_counter=224617311472, address=0, data=0),
OutputMessage(channel=9, timestamp=224617565634, rtio_counter=224617311582, address=0, data=0),
OutputMessage(channel=12, timestamp=224617565634, rtio_counter=224617311698, address=0, data=0),
OutputMessage(channel=15, timestamp=224617565634, rtio_counter=224617311796, address=0, data=0),
OutputMessage(channel=10, timestamp=224617565634, rtio_counter=224617311902, address=0, data=0),
OutputMessage(channel=13, timestamp=224617565634, rtio_counter=224617311990, address=0, data=0),
OutputMessage(channel=8, timestamp=224617565634, rtio_counter=224617312106, address=0, data=0),
OutputMessage(channel=11, timestamp=224617565634, rtio_counter=224617312196, address=0, data=0),
OutputMessage(channel=7, timestamp=224617565634, rtio_counter=224617312312, address=0, data=0),
]

compensated_ts = [evt.timestamp + compensation[evt.channel]
                  for evt in evts]
start_time = min(compensated_ts)
for ts in compensated_ts:
    print(ts-start_time)
```

which produces this output - it fits in the available lanes:

```
0 # 0
3 # 0
6 # 0
9 # 0
12 # 0
15 # 0
18 # 0
51 # 0
21 # 1
21 # 2
51 # 2
21 # 3
21 # 4
23 # 4
23 # 5
71 # 5
```

closed 2018-08-03 by sbourdeauducq