m-labs/artiq #1001: SERWB: use EN_VTC on Ultrascale

enjoy-digital, 2018-05-11:
IDELAYE3 needs to be in "TIME" mode. After a quick test, FPGA seems to lock at startup if mode="TIME" and EN_VTC=1. Need to investigate.

enjoy-digital, 2018-07-02:
I still need to look at that. But this only used for the high-speed phy and we are currently using low-spee phy.

hartytp, 2018-07-02:
@enjoy-digital remind me though, the plan is to move back to the high-speed phy for hw v2.0, right?

Is it worth adding some termination resistors on to the boards so that we can use the high-speed phy again with a supported IO standard?

enjoy-digital, 2018-07-02:
@hartytp: i'm not the one deciding, but if we can use LVDS on v2.0, yes we should use high-speed phy.

hartytp, 2018-07-02:
okay. Well, @sbourdeauducq let me know what you want to do about this then.

sbourdeauducq, 2018-07-02:
@enjoy-digital How does the low-speed PHY meet timing? Assume that with low bit rates and fast flip-flops, the chance of not meeting setup/hold is small?

sbourdeauducq, 2018-07-02:
> Is it worth adding some termination resistors on to the boards so that we can use the high-speed phy again with a supported IO standard?

I don't think faster programming of the DAC/HMC SPI etc. is that important and worth the trouble of reworking all boards (since otherwise, the master gateware would only work with some).
We should just add the resistors on the next board revision and switch to LVDS then.

enjoy-digital, 2018-07-02:
@sbourdeauducq: yes, we are:
1) Shifting clock sent by Master to the Slave by 180°.
2) Master outputs on rising edge, Slave samples on its rising-edge (falling-edge at Master).
3) Slave outputs on its rising edge (falling-edge at Master), Master samples on its rising-edge.

hartytp, 2018-07-02:
> I don't think faster programming of the DAC/HMC SPI etc. is that important and worth the trouble of reworking all boards (since otherwise, the master gateware would only work with some).
We should just add the resistors on the next board revision and switch to LVDS then.

No, it's not that important.

Just a matter of making sure we do as much testing as possible before sending Sayma v2.0 off to manufacture. We had a lot of serwb issues before, so it would be nice to see it working robustly at full speed on all boards before we consider it finished.

gkasprow, 2018-07-02:
At the moment 0201 resistors can be added between vias under the FPGA. But this is surgical operation under microscope.

sbourdeauducq, 2019-04-19:
Will keep the low-speed PHY on Sayma v2, and then move to pure DRTIO, so this becomes irrelevant.

closed 2019-04-19 by sbourdeauducq