m-labs/artiq #999: Kasli DRTIO satellite: link does not come up

cjbe, 2018-05-11:
With the current gateware the Kasli DRTIO satellite does not ever get to the 'link up' state - it hangs after "Si5324 is locked".

With the Kasli satellite running 2ca3fda898cb924c78b28ff179185a49b2553e71 the DRTIO link comes up properly, running dd21c07b85c1067913d521b09fda80acf561df7d or later and it hangs.
This happens independent of which version the master is running (i.e. the with the master at HEAD, and using the old satellite gateware, everything works fine).

This seems likely to be due to the clocking changes in misoc between 103bb3a8 and 4c010df2 .

marmeladapk, 2018-05-11:
It may be connected with what I saw - on some variants of Kasli I had rolling restarts of Artiq (but not misoc) once I got `Si5324 is locked`. Next week I'll investigate it more.

sbourdeauducq, 2018-05-12:
@cjbe Can you test again?

cjbe, 2018-05-12:
@sbourdeauducq that fixed it

closed 2018-05-12 by cjbe