m-labs/artiq #1004: Build Kasli CI, write tests, maintain

jordens, 2018-05-14:
* With the entire set of EEMs
* Comprehensive testing of PHYs
  * TTL in
  * TTL out
  * Freq gen
  * Counter
  * Zotino
  * Sampler
  * Urukul
  * SU-Servo
* Maybe with satellite/master setup as well

sbourdeauducq, 2018-08-17:
TTL in and out done.

sbourdeauducq, 2018-08-17:
> Freq gen (port)
> Zotino (to Sampler)
> Sampler (from Zotino, from Urukul)
> Urukul (to Sampler)

I have the hardware for those, all needs to be done is connect some SMA/BNC cables and write the tests. @jordens do you want to do the latter?

> Maybe with satellite/master setup as well

I have a second Kasli (1.0) as well, I can put it in the crate and connect its SFP to the main Kasli. Adding a pair of TTLs (one in each direction) connected to the main Kasli would go a long way to test DRTIO.

Another problem will be building the bitstreams in parallel to speed up tests.

sbourdeauducq, 2019-03-11:
I'm going to add a AD9912 Urukul to the CI setup, there have been a few frustrating issues with it lately.

sbourdeauducq, 2019-03-11:
Current status of tests:
* [X] I2C
* [X] TTL out -> in loopback
* [ ]  TTL Freq gen -> in
* [ ] Zotino (to Sampler)
* [ ] Sampler (from Zotino, from Urukul)
* [X] Urukul AD9910 (without observation of RF output)
* [ ] Urukul AD9910 (to Sampler)
* [ ] Urukul AD9912 (without observation of RF output)
* [ ] Urukul AD9912 (to Sampler)
* [ ] SU-Servo (Diode)
* [ ] DRTIO: not done
* [ ] parallel bitstream builds - works on nixbld, need to connect Kasli to it