m-labs/artiq #1005: KeyError in compiler when overloading methods

jordens, 2018-05-16:
```python
from artiq.experiment import *

class C:
    @kernel
    def f(self):
        pass

class D(C):
    @kernel
    def f(self):
        # super().f()  # super() not bound
        C.f(self)  # KeyError in compile

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.d = D()

    @kernel
    def run(self):
        self.d.f()
```

```

Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 219, in main
    return run(with_file=True)
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 205, in run
    raise exn
  File "/home/rj/src/artiq/artiq/frontend/artiq_run.py", line 198, in run
    exp_inst.run()
  File "/home/rj/src/artiq/artiq/language/core.py", line 53, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/src/artiq/artiq/coredevice/core.py", line 128, in run
    self.compile(function, args, kwargs, set_result)
  File "/home/rj/src/artiq/artiq/coredevice/core.py", line 111, in compile
    library = target.compile_and_link([module])
  File "/home/rj/src/artiq/artiq/compiler/targets.py", line 184, in compile_and_link
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/rj/src/artiq/artiq/compiler/targets.py", line 184, in <listcomp>
    return self.link([self.assemble(self.compile(module)) for module in modules])
  File "/home/rj/src/artiq/artiq/compiler/targets.py", line 138, in compile
    llmod = module.build_llvm_ir(self)
  File "/home/rj/src/artiq/artiq/compiler/module.py", line 89, in build_llvm_ir
    attribute_writeback=self.attribute_writeback)
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 491, in process
    self.process_function(func)
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 653, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1566, in process_Quote
    return self._quote(insn.value, insn.type, lambda: [repr(insn.value)])
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1507, in _quote
    return _quote_attributes()
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1489, in _quote_attributes
    lambda: path() + [attr])
  File "/home/rj/src/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 1553, in _quote
    self.embedding_map.retrieve_function(value))
  File "/home/rj/src/artiq/artiq/compiler/embedding.py", line 101, in retrieve_function
    return self.function_map[function]
KeyError: <artiq.compiler.embedding.SpecializedFunction object at 0x7f78ac3ab160>
```

* current master

sbourdeauducq, 2018-05-16:
Can we just have a better error message?

whitequark, 2018-05-16:
Sure. You don't have to repeat this every time there's an issue in the compiler. You know I take error messages very seriously and fix every instance.

sbourdeauducq, 2018-05-16:
Note the "just" - I was proposing only improving the error message and not implementing overloading.

whitequark, 2018-05-16:
I don't remember the reason for not having overloading. It might be that once I fix the bug, it'll just work, with the caveat above (which already kicks in under different circumstances).

whitequark, 2018-05-25:
Your example does in fact work after the fix, enjoy. Though you wouldn't be able to call `C.f` directly in more than one subclass, so the utility is not very high.

commit 12d1b981 (2018-05-25): compiler: handle direct calls to class methods.

Fixes #1005.
commit d0e5aec8 (2018-05-25): compiler: handle direct calls to class methods.

Fixes #1005.
closed 2018-05-25 by whitequark