m-labs/artiq #1011: missing  llvmlite-artiq 0.23.0.dev py35_4

jordens, 2018-05-23:
llvmlite-artiq 0.23.0.dev py35_4 doesn't exist for win64 making artiq uninstallable

jordens, 2018-05-23:
AFAICT I triggered the build, it was built and uploaded but now it doesn't exist anymore.
http://buildbot.m-labs.hk/builders/conda-win64/builds/250

jordens, 2018-05-23:
probably hit the conda repo size limit. i deleted a bunch of packages on anaconda.

jordens, 2018-05-23:
Was looking at the wrong package. Uploads still disappearing.

jordens, 2018-05-23:
Uploaded manually but the problem persists.

sbourdeauducq, 2018-05-26:
Since there is a workaround, it's not critical.

jordens, 2018-05-26:
That's a bad idea.

whitequark, 2018-05-26:
In the past it was solved by `conda update anaconda-client`.

jordens, 2018-05-26:
Pretty sure I had tried that on the buildbot.

whitequark, 2018-05-26:
No idea then. It also could be that the uploads failed at random, anaconda often has network issues, and by the time you manually uploaded the package that stopped.

jordens, 2018-05-26:
I probably did five attempts at automatic uploading.

sbourdeauducq, 2021-08-05:
Hydra + self-hosted conda channel does not have such issues.

closed 2021-08-05 by sbourdeauducq