m-labs/artiq #1015: SUServo: document connections/settings

hartytp, 2018-05-28:
It would be helpful to document:

- On Urukul DIP switch 2 (hw is not zero-indexed) must be ON to enable SUServo. 
- Urukul DIP switch 1 also needs to be on for HW v 1.0 to select the AD9910 variant
- Less important: Urukul and Sampler also need to be connected with 2 EEM connectors (QSPI mode). 
- IMHO, it's not worth documenting that a clock is required for Urkul (like saying Kasli needs power). Might be worth commenting that the servo itself is agnostic about the clock frequency.

jbqubit, 2018-05-30:
> IMHO, it's not worth documenting that a clock is required for Urkul

To a new user it may not be clear if clock is derived from DRTIO or external or from XO. Never hurts to reiterate frequency range and power.

hartytp, 2018-05-31:
> To a new user it may not be clear if clock is derived from DRTIO or external or from XO. Never hurts to reiterate frequency range and power.

Right. I'm not saying that it's a bad thing to document this -- in the same way that we should definitely document Kasli's power supply requirements properly at some point -- just that it's not a SU-servo issue, but rather a more general Urukul issue. IMO, the SU-servo documentation should say "these are the servo-specific settings, everything else is a standard Urukul setup." I don't consider fully documenting Urukul to be within the scope of the SU-servo project; it's a separate project.

jbqubit, 2018-05-31:
That makes sense. Keep the SU-servo docs to the diffs.

commit e1b0fcc2 (2018-05-31): suservo: add documentation on settings and setup

close #1015
closed 2018-05-31 by jordens