m-labs/artiq #1019: Sayma AD9154 config attempt failed

jbqubit, 2018-05-30:
I've tried several builds and only get partial (~50%) success booting Sayma. This is using Sayma_AMC (TS190717-6) sent to Duke from HK -- now temporarily being testing at UMD. Builds I've tested include db4d1878 and bcb9c3. Observations:

- after power-cycling and loading .bit from JTAG, success at fully booting is about 50%
- after re-loading  .bit from JTAG, success at fully booting is  <<50%

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1058.gbcb9c3d0
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011001100000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011001100000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 92 102 123 120 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111001110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 184+-78 175+-84 155+-77 143+-85 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003886s]  INFO(runtime): software version 4.0.dev+1058.gbcb9c3d0
[     0.010236s]  INFO(runtime): gateware version 4.0.dev+1058.gbcb9c3d0
[     0.016616s]  INFO(runtime): log level set to INFO by default
[     0.022321s]  INFO(runtime): UART log level set to INFO by default
[     0.028459s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062139s]  INFO(board_artiq::serwb): done.
[     0.065175s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547217s]  INFO(board_artiq::serwb): 0 errors
[     1.550524s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032573s]  INFO(board_artiq::serwb): 0 errors
[     3.035877s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967889s]  INFO(board_artiq::serwb): 0 errors
[     4.971202s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976749s]  INFO(board_artiq::serwb):   bitslip: 18
[     4.981786s]  INFO(board_artiq::serwb):   ready: 1
[     4.986562s]  INFO(board_artiq::serwb):   error: 0
[     4.991337s]  INFO(board_artiq::serwb): RTM serwb settings:
[     4.996902s]  INFO(board_artiq::serwb):   bitslip: 25
[     5.001939s]  INFO(board_artiq::serwb):   ready: 1
[     5.006715s]  INFO(board_artiq::serwb):   error: 0
[     5.011725s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1058.gbcb9c3d0
[     5.019129s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.019006s]  INFO(runtime): continuing boot
[     6.022067s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     6.028307s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     6.034474s]  INFO(board_artiq::hmc830_7043::hmc7043): shutting down
[     6.040829s]  INFO(board_artiq::hmc830_7043::hmc830): loading configuration...
[     6.048321s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     6.053909s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     6.060714s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[     6.066775s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...
[     6.076019s]  INFO(board_artiq::hmc830_7043::hmc7043):   ...done
[     6.081076s]  INFO(board_artiq::ad9154): AD9154-0 found
[     6.095967s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.108900s]  INFO(board_artiq::ad9154):   ...done
[     6.373010s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #0 failed (JESD ready timeout), retrying
[     6.391342s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.399894s]  INFO(board_artiq::ad9154):   ...done
[     6.664005s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #1 failed (JESD ready timeout), retrying
[     6.682333s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.690882s]  INFO(board_artiq::ad9154):   ...done
[     6.955005s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #2 failed (JESD ready timeout), retrying
[     6.973333s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.981882s]  INFO(board_artiq::ad9154):   ...done
[     7.246006s]  WARN(board_artiq::ad9154): AD9154-0 config attempt #3 failed (JESD ready timeout), retrying
[     7.264335s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
```

jbqubit, 2018-05-30:
Similar behavior from 8fd57e6ccb6e5070d3dc988a9d5550b295621d72 which is yesterday's master.

sbourdeauducq, 2018-05-30:
Before https://github.com/m-labs/jesd204b/commit/03718be2f697ea928766342d16952916c20ee1e6 JESD was intermittently failing to initialize on all boards.

jbqubit, 2018-05-30:
This is the JESD I'm using.

```
$ conda list jesd204b
# packages in environment at /home/britton/miniconda3/envs/artiq-dev:
#
jesd204b                  0.5                        py_1    m-labs/label/main
```

Please package the right version of jesd204b with conda. 

@sbourdeauducq Do you have updated installation instructions for Sayma standalone? [This post](https://ssl.serverraum.org/lists-archive/artiq/2018-January/001176.html) is the most recent that I know about. Diffs presently are the following.
- no HMC830 patch -- HMC830 locks!
- apply 100 MHz (+10 dBm) to SMA clock input on RTM
- ```conda install --force jesd204b=0.6```

[This](https://github.com/jbqubit/sinara-testing/blob/master/sayma/tools/build-artiq.bash) is the script I'm using to automate my builds.

jbqubit, 2018-05-30:
Building master with jesd204b=0.6 now.

jbqubit, 2018-05-30:
We updated JESD and built from master 38b51282226f9. We reloaded RTM AMC .bit and observed successful booting each time. This looks resolved.

jbqubit, 2018-05-30:
But spoke too soon. Built with 8fd57e6ccb6e (and JESD update) and find that the same error happens. ''' AD9154-0 initializing...'''

gkasprow, 2018-05-30:
Unfortunately we have a holidays and the university is closed. Would be able to check it on Monday...

jbqubit, 2018-05-30:
Ignore that post from five minutes ago. I accidentally used a build of the gateware from this morning that predated the JESD204B upgrade to v0.6. Just now I repeated test again using JESD204B=0.6 with SAWG and HMC830 build 38b51282226f9. Boot was successful four times in succession.