m-labs/artiq #1020: Sayma boot hangs at hmc7043

jbqubit, 2018-05-30:
I've tried several builds and only get partial (~50%) success booting Sayma. This is using Sayma_AMC (TS190717-6) sent to Duke from HK -- now temporarily being testing at UMD. Builds I've tested include db4d1878 and bcb9c3. Observations:

- after power-cycling and loading .bit from JTAG, success at fully booting is about 50%
- after re-loading  .bit from JTAG, success at fully booting is  <<50%

Common failure is hanging at loading configuration of ```hmc7043```. Sometimes this happens on first boot attempt. Sometimes it happens after automatic restart after panic. 

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1058.gbcb9c3d0
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011001100000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011001100000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001010111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 92 102 123 120 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110101111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111001110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 184+-78 175+-84 155+-77 143+-85 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003886s]  INFO(runtime): software version 4.0.dev+1058.gbcb9c3d0
[     0.010236s]  INFO(runtime): gateware version 4.0.dev+1058.gbcb9c3d0
[     0.016616s]  INFO(runtime): log level set to INFO by default
[     0.022321s]  INFO(runtime): UART log level set to INFO by default
[     0.028459s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062139s]  INFO(board_artiq::serwb): done.
[     0.065175s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547217s]  INFO(board_artiq::serwb): 0 errors
[     1.550524s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032573s]  INFO(board_artiq::serwb): 0 errors
[     3.035877s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967889s]  INFO(board_artiq::serwb): 0 errors
[     4.971202s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976749s]  INFO(board_artiq::serwb):   bitslip: 18
[     4.981786s]  INFO(board_artiq::serwb):   ready: 1
[     4.986562s]  INFO(board_artiq::serwb):   error: 0
[     4.991337s]  INFO(board_artiq::serwb): RTM serwb settings:
[     4.996902s]  INFO(board_artiq::serwb):   bitslip: 25
[     5.001939s]  INFO(board_artiq::serwb):   ready: 1
[     5.006715s]  INFO(board_artiq::serwb):   error: 0
[     5.011725s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1058.gbcb9c3d0
[     5.019129s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[... 40 lines cut ...]
[     0.044201s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.526244s]  INFO(board_artiq::serwb): 0 errors
[     1.529550s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.011600s]  INFO(board_artiq::serwb): 0 errors
[     3.014904s]  INFO(board_artiq::serwb): Wishbone test...
[     4.989420s]  INFO(board_artiq::serwb): 0 errors
[     4.992733s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.998280s]  INFO(board_artiq::serwb):   bitslip: 3
[     5.003229s]  INFO(board_artiq::serwb):   ready: 1
[     5.008006s]  INFO(board_artiq::serwb):   error: 0
[     5.012781s]  INFO(board_artiq::serwb): RTM serwb settings:
[     5.018347s]  INFO(board_artiq::serwb):   bitslip: 0
[     5.023296s]  INFO(board_artiq::serwb):   ready: 1
[     5.028072s]  INFO(board_artiq::serwb):   error: 0
[     5.033089s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1058.gbcb9c3d0
[     5.040486s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.040008s]  INFO(runtime): continuing boot
[     6.043071s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     6.049314s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     6.055476s]  INFO(board_artiq::hmc830_7043::hmc7043): shutting down
[     6.061832s]  INFO(board_artiq::hmc830_7043::hmc830): loading configuration...
[     6.069334s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     6.074911s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     6.081717s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[     6.087777s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...

jbqubit, 2018-05-30:
As in https://github.com/m-labs/artiq/issues/1019
we updated JESD and built from master 38b51282226f9. We reloaded RTM AMC .bit four times and observed successful booting each time. This looks resolved.

sbourdeauducq, 2018-05-30:
The JESD core does not touch the HMC7043. This looks like typical Sayma insane behavior.

hartytp, 2018-05-31:
@sbourdeauducq is there any way to distinguish ser-wb lock-ups from, say, some memory corruption issue on the AMC due to some SDRAM issue?

sbourdeauducq, 2018-05-31:
Maybe observe the serwb signals with microscope after the lockup.

sbourdeauducq, 2018-05-31:
Though it can be anything, not necessarily serwb or the SDRAM...

enjoy-digital, 2018-05-31:
@hartytp: if you are able to reproduce the hangs/crashes, i first suggest to verify the ouptuts of the hmc830 and hmc7043. We already saw that it can cause some broadband noise and having a crash just when we are configuring and enabling the outputs of the hmc7043 seems strange... It also strange that when we have the first crashes, we are then not able to pass the hmc7043 configuration anymore.

Maybe some component is heating up (hmc830 or hmc7043?), then generates noise and causes the crashes.

So the first thing to look at:
- add some debug info to hmc7043 and slow down the configuration to be able to have the traces.
- try to see if it's always at the same register configuration.
- try to see if you still have the crashes by commenting https://github.com/m-labs/artiq/blob/master/artiq/firmware/libboard_artiq/ad9154.rs#L665. (reset is also controling the clock input buffers of the FPGA). If it's no longer crashing, then it means hmc7043 is still generating noise.

jbqubit, 2018-05-31:
> do you get garbage out of the UART or illegal instruction errors

No UART garbage or illegal instruction errors were observed. 

Continuing to use 38b51282 today with HMC830 and updated JESD204b. Haven't observed this hang again. Will post if I do.

hartytp, 2018-05-31:
@enjoy-digital ack. If I start to be able to reproduce the issue then I'll have a look at that. I'll also look at the 7043 outputs on a fast scope and see how they look.

@gkasprow it may be worth considering adding a FET to disable the HMC7043 output bias until after everything is locked and properly configured.

jbqubit, 2018-05-31:
@hartytp When will a Sayma arrive in your lab?

hartytp, 2018-05-31:
it was posted yesterday. Still being processed by La Poste atm

jbqubit, 2018-05-31:
> it may be worth considering adding a FET to disable the HMC7043 output bias until after everything is locked and properly configured.

It looks like each channel on the the HMC7043 has an enable bit (LSB of CSR in [pg 30](http://www.analog.com/media/en/technical-documentation/data-sheets/HMC7043.pdf)). That register's default value is unfortunately ```1```. Are there other output enable/disable toggles on HMC7043 that could block output until it's properly configured?

hartytp, 2018-05-31:
>  Are there other output enable/disable toggles on HMC7043 that could block output until it's properly configured?

Not as far as I'm aware, which is why I suggested the FETs...

It might be that muting the outputs early on in the boot is enough, but obviously that won't protect us during the initial booting.

jbqubit, 2018-05-31:
Being able to constrain output of RF system to only "good behavior" is really important. Even when booting.  One manifestation of this is #1031.

sbourdeauducq, 2018-05-31:
> One manifestation of this is #1031.

That's just speculation.

hartytp, 2018-05-31:
@jboulder during booting the AFE RF switches will be close, so that's not an issue.

hartytp, 2018-06-05:
Let's close this, as I don't think it's happening any more after the RESET fix.