m-labs/artiq #1025: Sayma ping >> 30 ms

jbqubit, 2018-05-30:
Using build from latest master. Usually ping is < 1 ms. Lately ping has been much longer.

```
--- 192.168.1.75 ping statistics ---
69 packets transmitted, 69 received, 0% packet loss, time 68077ms
rtt min/avg/max/mdev = 32.764/33.258/43.353/1.676 ms
```

This seems to be the case for builds I've tested in the last week. It does not seem to be a function of switch (I've tried two). It's not due to other network traffic (one of the switches only has two devices).

trxw, 2018-05-31:
net_trace slows down ping response. this makes sense now. closing. @sbourdeauducq 

```artiq_mkfs -s net_trace 1 -s panic_reset 1 -s mac 00:0a:35:03:1e:75 -s ip 192.168.1.75 -f startup_kernel startup_kernel.elf sayma.config```

```
$ ping 192.168.1.75
PING 192.168.1.75 (192.168.1.75) 56(84) bytes of data.
64 bytes from 192.168.1.75: icmp_seq=1 ttl=64 time=32.8 ms
64 bytes from 192.168.1.75: icmp_seq=2 ttl=64 time=32.8 ms
^C
--- 192.168.1.75 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1002ms
rtt min/avg/max/mdev = 32.841/32.851/32.862/0.181 ms
```

---------------

```artiq_mkfs -s net_trace 0 -s panic_reset 1 -s mac 00:0a:35:03:1e:75 -s ip 192.168.1.75 -f startup_kernel startup_kernel.elf sayma.config```

```
$ ping 192.168.1.75
PING 192.168.1.75 (192.168.1.75) 56(84) bytes of data.
64 bytes from 192.168.1.75: icmp_seq=1 ttl=64 time=0.361 ms
64 bytes from 192.168.1.75: icmp_seq=2 ttl=64 time=0.328 ms
64 bytes from 192.168.1.75: icmp_seq=3 ttl=64 time=0.297 ms
64 bytes from 192.168.1.75: icmp_seq=4 ttl=64 time=0.358 ms
```

trxw, 2018-05-31:
This was @jboulder. Just noticed that @trxw was logged into github. I'll log him out for further posts.

whitequark, 2018-05-31:
@sbourdeauducq net_trace is currently not documented, should we document it and add that caveat?

sbourdeauducq, 2018-05-31:
Sure, go ahead.

jbqubit, 2018-05-31:
Great! Thank you @whitequark. :)