m-labs/artiq #1027: SU-servo: clarify timings

hartytp, 2018-05-31:
Please can you clarify the timing of the update cycles for the servo a bit in the docs? Am I right in thinking that:
1. When the servo is enabled using set_config, the update cycle will start immediately (current now()). The first DDS update will thus be just less than 1 servo cycle (1.1us) afterwards. The delay between the first ADC sample/DDS update and enabling the servo is constant.
2. So, to synchronize the servo operations with an experiment, the servo should be disabled and re-enabled.

hartytp, 2018-05-31:
4. The delay set using `set_iir` is rounded to the nearest number of 1.2us servo cycles. So, e.g., a 3 us delay will round up to about 3.5us. This delay is effectively a lower bound on the delay, since it is synchronous with the servo update timings. So, a delay of 3us means that the actual delay will lie between 3.5us and 4.7us.
5. The delay is the delay before the IIR begins operating. e.g. it's roughly the delay before the first ADC sample is taken. DDS updates occur at the end of the servo cycle, so a 3us delay means that the first DDS update will occur between about 4.7 and 6us after the RF switch opens.

hartytp, 2018-05-31:
Side note: would it make more sense to rename `set_config` to `set_enable` since that's all it does atm?

jordens, 2018-05-31:
Ack. A couple comments:

1. The first DDS update after enabling the servo globally is about 2 servo cycles later (the pipeline is empty and ADC/IIR are roughly one cycle and DDS is another).
2. We can add a feature that gives you synchronization on-the-fly by supporting emission of an RTIO input event when a cycle starts. The other way to synchronize is to calculate the cycle restart from the known enable timestamp and the cycle time.
3. No. Enabling a channel behaves like a regular TTL channel ("instantaneous", in parallel, replacement supported) with a bit more features (servo profile, delay). There is no memory access involved.
4. Yes.
5. The changed rf switch and profile settings of a channel are taken into account during the IIR stage.
6. I wouldn't want to do that because it's likely to need reverting. As soon as e.g. (2) is implemented we'd have to change it again.

hartytp, 2018-05-31:
> I wouldn't want to do that because it's likely to need reverting. As soon as e.g. (2) is implemented we'd have to change it again.

ack. If you think we're likely to add more config settings in the future then I agree the current naming is correct.

hartytp, 2018-05-31:
> We can add a feature that gives you synchronization on-the-fly by supporting emission of an RTIO input event when a cycle starts. The other way to synchronize is to calculate the cycle restart from the known enable timestamp and the cycle time.

Personally, I don't think this is needed (@cjbe). Spelling this out a little more in the docs is fine IMHO.

commit 22506e84 (2018-06-01): suservo: clarify timings

close #1027
closed 2018-06-01 by jordens