m-labs/artiq #1029: artiq_rpctool does not cope with kwonlyargs

jordens, 2018-05-31:
This is `artiq_rpctool ... list-methods` on the `newfocus8742` controller.

```
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_rpctool", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_rpctool')()
  File "/home/rj/src/artiq/artiq/frontend/artiq_rpctool.py", line 138, in main
    list_methods(remote)
  File "/home/rj/src/artiq/artiq/frontend/artiq_rpctool.py", line 69, in list_methods
    if kwonlyarg in argspec["kwonlydefaults"]:
TypeError: argument of type 'NoneType' is not iterable
```

The signature is: `f(self, xx=None, *nn)`
   
`argspec` is: `{'varargs': 'nn', 'kwonlydefaults': None, 'varkw': None, 'defaults': None, 'args': ['self'], 'annotations': {}, 'kwonlyargs': ['xx']}`

whitequark, 2018-06-01:
No, the signature is: `do_dh(self, *nn, xx)`. You have the wrong method, `f` is fine.

jordens, 2018-06-01:
Right! Sorry for the red herring.

commit 985fd737 (2018-06-01): artiq_rpctool: use inspect.formatargspec instead of a NIH formatter.

Fixes #1029.
commit 408734b7 (2018-06-11): artiq_rpctool: use inspect.formatargspec instead of a NIH formatter.

Fixes #1029.
closed 2018-06-01 by whitequark