m-labs/artiq #1031: Sayma spiky output when DACS are not configured

jbqubit, 2018-05-31:
Sayma produces spikes on RF output under several circumstances. Some circumstances where this is reproducibly seen are. 
- after kernel panic https://github.com/m-labs/artiq/issues/1026
- during boot from MISOC banner to about when DRAM is initialized
- on properly booted board after running ```artiq_flash -t sayma -f sayma.config storage ```

See attached screen shot. AC coupled looking at SMPs. DAC outputs:
- yellow: J16
- blue: J49
- red: J46     
- green: J19

[image]

sbourdeauducq, 2018-05-31:
Another consequence of the HMC7043 noise?

hartytp, 2018-05-31:
If you really want to get rid of this, we'd probably have to in a hardware AND with a JESD link up symbol or something like that.

Generally though, I have to say that it's almost impossible to guarantee that this kind of glitch can never happen on the outputs. It's much better to rely on good design to ensure that no DAC outputs can cause damage to the experiment (e.g. by setting the step attenuators correctly). More generally, anyone with very stringent interlocking requirements should consider using some extra hw outside Sayma to interlock it (cf booster).

sbourdeauducq, 2018-05-31:
Is that happening with a freshly powered system (nothing in RTM FPGA) with:
* a dummy bitstream in the AMC (e.g. LED blinker gateware) 
* nothing at all in the AMC

sbourdeauducq, 2018-05-31:
And with the current code, the RF switches are open as long as the RTM FPGA is loaded - it's just hardcoded like this.