m-labs/artiq #1034: suservo: offset=1

hartytp, 2018-05-31:
In a setup where the max ADC value can never be reached (DDS railing gives about -0.5FS) setting `offset=0.99` puts the DDS amplitude to max, while setting `offset=1` puts it to zero.

I guess this is a standard issue, but it should at least be documented (otherwise, setting offset=-1 is a pretty standard thing to do!).

jordens, 2018-05-31:
Yeah. it's two's complement where the positive max is one LSB smaller than the negative. Will document.

jordens, 2018-05-31:
Hmm. On re-reading I don't understand the first paragraph. Just to clarify: This is a setup with a positive ADC working point and thus negative `offset`. Do you mean `offset=-0.99` and `offset=-1` in the first paragraph? Because those two should be fine.

commit a42f7744 (2018-05-31): suservo: document offset range limits

close #1034
closed 2018-05-31 by jordens