m-labs/artiq #1036: Sayma SAWG test suite

jbqubit, 2018-05-31:
@sbourdeauducq I wrote a suite of tests that were originally used on KC705 with SAWG. This seems like a good starting point for testing SAWG on Sayma. They're located on github [here](https://github.com/jbqubit/sinara-testing/blob/master/sayma/repository/sawg-tests/test_suite.py).

I started running through the tests and get unexpected output for the simplest ```test_simple_sin1.py```. Embedded in each file are instructions on how to setup your scope and the expected results. Please do these tests as this is baseline for how SAWG is to perform on Sayma.

sbourdeauducq, 2018-05-31:
The SAWG gateware is essentially the same on KC705-phaser and on Sayma. So, you're probably seeing general Sayma bugginess and nothing that is SAWG specific.

sbourdeauducq, 2018-05-31:
Also, "unexpected output" is nothing I can help you with.

jbqubit, 2018-05-31:
When I tested SAWG on KC705 in summer 2017 I found quite a few bugs. @jordens worked with me to resolve many of them but the process was painful. The point of creating a test suite is to codify in code my baseline expectations. They're simple tests you can run on your own hardware. And, there's no communication delay or inevitable misunderstands that arise in human language. Please give it a shot. 

> Also, "unexpected output" is nothing I can help you with.

Read the source. ```test_simple_sin1.py``` tells you what the expected outcome is. I see noise which differs from ```expectation```. 

```test_simple_sin1
    purpose: test sinusoidal output
    expectation:
      - 10 MHz from sawg.frequency1 on sawg0 and sawg1
      - playback continues after program termination
    setup: scope single run, force trigger manually, 500ns/div, 100mV/div
```

closed 2018-06-01 by sbourdeauducq