m-labs/artiq #1040: Sayma SAWG frequency1/frequency2 generators produce broken waveform

sbourdeauducq, 2018-06-01:
```python
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("sawg1")

    @kernel
    def run(self):
        self.core.reset()
        self.sawg1.reset()
        delay(300 * us)

        self.sawg1.frequency1.set(10*MHz)
        self.sawg1.amplitude1.set(0.5)
```

[image]

As mentioned in #1039, using ``frequency0`` instead produces a more reasonable waveform.

jordens, 2018-06-01:
Could you elaborate on the difference to #1039 and #1022? That same test code works fine in #1022 .
And are both duplicates of #1022 ?

sbourdeauducq, 2018-06-01:
> That same test code works fine in #1022

No, #1022 uses frequency0. The bug described here appears with frequency1.

jbqubit, 2018-06-01:
Sorry. I created similar bug report before seeing this one. #1040

hartytp, 2018-06-01:
@jbqubit cf https://irclog.whitequark.org/m-labs/2018-06-01

hartytp, 2018-06-05:
@jbqubit @sbourdeauducq and anyone else who is working with Sayma v1.0: I strongly suggest that you reconnect the HMC7043 RESET line before doing any further debugging. We've seen that not holding the HMC7043 in RESET during init caused crashes on two separate boards (mine and @marmeladapk's). It's quite possible that other boards are affected in different ways, even if they don't crash.

gkasprow, 2018-06-05:
one can reconnect it but either negator is needed or one need to connect it to one of IO pins.

jbqubit, 2018-06-05:
As @gkasprow said, looks like either HMC743 or Si5324 would be held in reset as they share the signal but have inverted reset logic.

klickverbot, 2018-06-05:
That is true, but for the current prototype phase, missing clock recovery is easier to tolerate than random crashes. A "proper" bodge might be preferable, as Greg points out, if we all agree on which pin to route the reset to avoid having to use different firmware builds.

klickverbot, 2018-06-05:
(Also see https://github.com/m-labs/artiq/commit/ac5c4913ec99579e8035276e0da5b2933e2d3722.)

sbourdeauducq, 2018-06-06:
That Si5324 is only intended for switched DRTIO on the RTM FPGA, i.e. rf switch control etc. -it's a minor  feature.
And, unlike the HMC CPU interfaces and behavior which are a disaster, the Si CPU interface is well though through. Case in point, the reset can be disconnected (like it's done on Kasli) on the Si5324 without consequences or obscure bugs popping up.

marmeladapk, 2018-06-06:
Reproduced:

[image]

Artiq: c28fe47, Misoc: 86c6fcb, Migen: 6425844, jesd204b: 25fd79d

closed 2018-06-08 by jordens

jbqubit, 2018-06-08:
@jordens Still broken using 4.0.dev+1132.g5b73dd86. Garbage output and kernel halt. 

```
[    38.610685s]  INFO(runtime::session): no idle kernel found
[    46.097919s]  INFO(runtime::session): new connection from 192.168.1.68:45592
[    46.145662s]  INFO(runtime::kern_hwreq): resetting RTIO
@ 0x40015ba0
+0000: 84b80004 
```

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]
        self.freq_list = self.linspace(50.0*MHz, 200.0*MHz, 1000)
        self.amp_list = self.linspace(0.0, 1.0, 100)

    def linspace(self, start, stop, num):
        dx = (stop-start)/(num-1) 
        return [start+n*dx for n in range(num)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)
        freq0 = 10*MHz
        amp1 = 0.5
        for sawg in self.sawgs:
            sawg.reset()
            delay(500*us)
            sawg.frequency0.set(freq0)
            sawg.amplitude1.set(amp1)

        while True:
            for x in self.freq_list:
                self.ttl_sma_out.pulse(1*us)
                for sawg in self.sawgs:
                    sawg.frequency1.set(x)
                    delay(1*ms)
```

sbourdeauducq, 2018-06-09:
Did you rework the hmc7043 reset and update misoc?

jbqubit, 2018-06-09:
> update misoc

Yes, misoc [matches](https://github.com/m-labs/artiq/commit/6a7983cf89a239706f0ebacd04e6b7ac8a82620b) that in updated YAML. 

```
(artiq-dev) britton@artiqbuild:~/artiq-dev/artiq$ conda list|grep misoc
misoc                     0.11            py35_20+git2436a68d    m-labs/label/dev
```

> rework the hmc7043 reset

Unclear as to what was settled upon as to the rework. 
- [no hardware mod](https://github.com/m-labs/artiq/pull/1049#issuecomment-394673745) 
- [pull up resistor](https://irclog.whitequark.org/m-labs/2018-06-07#22282008;) -- if so what resistance did you use to P3V3 @hartytp? 

In absence of this rework I don't see serwb or clock configuration errors during boot on my hardware. What modification did you make on HK Sayma? It's unclear from discussion on IRC if anyone has repeated tests of SAWG.

sbourdeauducq, 2018-06-10:
@jbqubit the HK board working again and the SAWG looks OK now, you should test.

trxw, 2018-06-12:
@jbqubit Is this issue resolved with the board at UMD?

jbqubit, 2018-06-13:
Yes. I've confirmed this at several specific frequencies using startup kernel. Kernel panics prevent sweeping across a range of frequencies so testing hasn't been exhausted. But please close this issue as I suspect kernel panics are a separate problem.

commit 6a7983cf (2018-06-08): conda: bump misoc

closes #1039
closes #1040
closes #1022
closes #1058
closes #1044