m-labs/artiq #1041: SUServo: boiler plate

hartytp, 2018-06-01:
At the moment, there is quite a lot of boiler plate required to add a SUServo to a build.

https://github.com/m-labs/artiq/blob/87d3ac9d250af838c7f8444185eeffcb9e8e6209/artiq/gateware/targets/kasli.py#L214-L274

I do not expect users to alter any servo parameters apart from `t_rtt`. So, I'd prefer not to expose them in the target file definitions.

Ideally, I'd really like to have a 1-line way of adding a SU-servo to a target, which takes only eem locations and a `t_rtt` its parameters. cf the nice EEM classes that have been introduced.

In the docstring for the `t_rtt` parameter, can we note that the `t_rtt >= 4 + round_trip_cable_delay/8ns` and that `t_rtt > ?` slows down the servo iterations.

Other than that, I'm happy with the SUServo interface.

jordens, 2018-06-01:
At least `shift`, `profile`, `clk`, are parameters worth changing as well. But the rest could be refactored into `eem` and `add_std()`

hartytp, 2018-06-01:
ack.

commit f50aef1a (2018-06-01): suservo: extract boilerplate

closes #1041
closed 2018-06-01 by jordens