m-labs/artiq #1043: Sayma kernel freeze without panic, random characters

jbqubit, 2018-06-01:
Using build 20180530 with SAWG 38b51282226f9. Now seeing failure different from #1026. Random characters printed on URART, kernel freeze but no kernel panic. @whitequark continuing conversation started on IRQ with @trxw. https://irclog.whitequark.org/m-labs/2018-06-01#22239059

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1087.g38b51282
Initializing SDRAM...
DQS initial delay: 102 taps
Write leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100011000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001100000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000101111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100011000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 102 taps
Write leveling: 98 103 129 121 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111101111111000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 185+-77 174+-85 154+-79 142+-81 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003886s]  INFO(runtime): software version 4.0.dev+1087.g38b51282
[     0.010236s]  INFO(runtime): gateware version 4.0.dev+1087.g38b51282
[     0.016611s]  INFO(runtime): log level set to INFO by default
[     0.022316s]  INFO(runtime): UART log level set to INFO by default
[     0.028459s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062139s]  INFO(board_artiq::serwb): done.
[     0.065175s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547217s]  INFO(board_artiq::serwb): 0 errors
[     1.550524s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032573s]  INFO(board_artiq::serwb): 0 errors
[     3.035877s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967817s]  INFO(board_artiq::serwb): 0 errors
[     4.971129s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976676s]  INFO(board_artiq::serwb):   bitslip: 39
[     4.981713s]  INFO(board_artiq::serwb):   ready: 1
[     4.986489s]  INFO(board_artiq::serwb):   error: 0
[     4.991264s]  INFO(board_artiq::serwb): RTM serwb settings:
[     4.996830s]  INFO(board_artiq::serwb):   bitslip: 4
[     5.001779s]  INFO(board_artiq::serwb):   ready: 1
[     5.006556s]  INFO(board_artiq::serwb):   error: 0
[     5.011565s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1087.g38b51282
[     5.018969s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.018006s]  INFO(runtime): continuing boot
[     6.021067s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     6.027308s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     6.033474s]  INFO(board_artiq::hmc830_7043::hmc7043): shutting down
[     6.039829s]  INFO(board_artiq::hmc830_7043::hmc830): loading configuration...
[... 59 lines cut ...]
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        for sawg in self.sawgs:
            sawg.reset()
            delay(1*us)
            sawg.amplitude1.set(.5)
            sawg.amplitude2.set(.5)

            sawg.frequency0.set(30*MHz)
            sawg.frequency1.set(30*MHz)
            sawg.frequency2.set(30*MHz)
            
            delay(100*ns)

        while True:
            delay(0.5*ms)
            self.ttl_sma_out.pulse(0.6*ms)
```

No response to ping.

jbqubit, 2018-06-01:
This time panic and garbage characters. Panic message looks different. 

```
 __  __ _ ____         ____ 
|  \/  (_) ___|  ___  / ___|
| |\/| | \___ \ / _ \| |    
| |  | | |___) | (_) | |___ 
|_|  |_|_|____/ \___/ \____|

MiSoC Bootloader
Copyright (c) 2017-2018 M-Labs Limited

Bootloader CRC passed
Gateware ident 4.0.dev+1087.g38b51282
Initializing SDRAM...
DQS initial delay: 101 taps
Write leveling scan:
Module 3:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111001100000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111000011000000000000000000000000000000000000000000000000000000000000000000
Module 1:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110111010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000110111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110110010000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
DQS initial delay: 101 taps
Write leveling: 94 105 128 122 done
Read leveling scan:
Module 3:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110011101110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 2:
00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111100111100000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 1:
00000000000000000000000000000000000000000000000000000000000000000000000011111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Module 0:
00000000000000000000000000000000000000000000000000000000100001111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111111011110000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
Read leveling: 188+-76 177+-85 155+-78 142+-82 done
SDRAM initialized
Memory test passed

Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003886s]  INFO(runtime): software version 4.0.dev+1087.g38b51282
[     0.010236s]  INFO(runtime): gateware version 4.0.dev+1087.g38b51282
[     0.016611s]  INFO(runtime): log level set to INFO by default
[     0.022316s]  INFO(runtime): UART log level set to INFO by default
[     0.028459s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062138s]  INFO(board_artiq::serwb): done.
[     0.065173s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547216s]  INFO(board_artiq::serwb): 0 errors
[     1.550523s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032572s]  INFO(board_artiq::serwb): 0 errors
[     3.035876s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967898s]  INFO(board_artiq::serwb): 0 errors
[     4.971210s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976758s]  INFO(board_artiq::serwb):   bitslip: 15
[     4.981795s]  INFO(board_artiq::serwb):   ready: 1
[     4.986570s]  INFO(board_artiq::serwb):   error: 0
[... cut ...]
[    31.929532s]  INFO(runtime::session): no idle kernel found
[    64.824224s]  INFO(runtime::session): new connection from 192.168.1.68:58344
[    64.868325s]  INFO(runtime::kern_hwreq): resetting RTIO
panic at runtime/main.rs:305:14: exception IllegalInsn at PC 0x4003c9c0, EA 0x40152bbc
backtrace for software version 4.0.dev+1087*g38b51282:
panic at runtime/main.rs:305:14: exception IllegalInsn at PC 0x40059738, EA 0x401503d0
backtrace for software version 4.0.dev+1087*g38b51282:
�
```
Same identical runtime.elf.

whitequark, 2018-06-01:
Disassembly:
```
4003c9c0:       d7 e2 1e f0     l.sw -272(r2),r3
40059738:       84 81 00 18     l.lwz r4,24(r1)
```
These are definitely legal. The fact that they're load/store instructions is probably just because the CPU spends most execution time spinning one of those these.

sbourdeauducq, 2018-06-01:
What vivado version are you using?

jbqubit, 2018-06-01:
I'm using 2017.4. What version of Vivado are you using?? https://github.com/m-labs/artiq/issues/910

jbqubit, 2018-06-01:
> ****** Vivado v2018.1 (64-bit)

[Says](http://buildbot.m-labs.hk/builders/artiq-board/builds/1605/steps/conda_build/logs/stdio) the build bot.

sbourdeauducq, 2018-06-03:
FWIW, with 2018.1 I've run two different Sayma boards (after the various fixes bugs like SDRAM, HMC7043 noise, 1V8, etc.) continuously for days without any bug of this sort.

hartytp, 2018-06-03:
I'm using Vivado 2018.1 (current master doesn't meet timing on the 2017 version I was using), and see occasional illegal instructions on boot.

sbourdeauducq, 2018-06-03:
Yes, we need the memory dump,the rest of the crash message, and the corresponding ``runtime.elf``.

hartytp, 2018-06-03:
> I'm using 2017.4. What version of Vivado are you using?? #910

Have you double checked that your build meets timing? It didn't for me on 2017.4...

sbourdeauducq, 2018-06-04:
Yes.

hartytp, 2018-06-04:
@sbourdeauducq my question was addressed to @jbqubit (see the text I quoted) as he is still using 2017.4, where Sayma didn't meet timing for me...

jbqubit, 2018-06-04:
> FWIW, with 2018.1 I've run two different Sayma boards (after the various fixes for bugs like SDRAM, HMC7043 noise, 1V8, etc.) continuously for days without any bug of this sort.

Typical use case for Sayma is running a variety of ARTIQ Python scripts interactively and scheduled that exercise a variety of PHYs. What does are you Sayma doing continuously during these tests?

sbourdeauducq, 2018-06-04:
Always running the same kernel that uses SAWG, but there was Ethernet traffic processed by the comms CPU due to TCP keepalive and network broadcasts.
Did you find some way to reproduce those crashes?

jbqubit, 2018-06-04:
I'll use 2018.1 going forward. --without-sawg builds with problem. Now building with SAWG.

I can't reproduce the crashes in this Issue as they are triggered by communication with the Core device via Ethernet. Ethernet is not working on my Sayma.

sbourdeauducq, 2018-06-04:
> --without-sawg builds with problem.

What problem? Can you open a new issue?

> Ethernet is not working on my Sayma.

It was working before. What broke it?

jbqubit, 2018-06-04:
> --without-sawg builds with problem. 

Sorry... Typo. --without-sawg build /without/ problem. --sheepish--

jbqubit, 2018-06-04:
Using latest from master 20180604 with SAWG vivado 2018.1 07d4145a35c739. Meets timing.  I've run 25 scripts involving SAWG via Ethernet. I don't see any errors on UART.

closed 2018-06-06 by jbqubit