m-labs/artiq #1044: Sayma SAWG frequency0.set(240*MHz) produces nonsense

jbqubit, 2018-06-01:
Sayma SAWG frequency0.set(240*MHz) produces nonsense. Frequencies below about 150*MHz produce sinusoids. 

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        for sawg in self.sawgs:
            sawg.reset()
            delay(1*us)
            sawg.amplitude1.set(.5)

            sawg.frequency0.set(240*MHz)
            
            delay(100*ns)

        while True:
            delay(0.5*ms)
            self.ttl_sma_out.pulse(0.6*ms)
```

[image]

jbqubit, 2018-06-04:
@sbourdeauducq Do you see this on your hardware?

sbourdeauducq, 2018-06-04:
As pointed out elsewhere by Tom and Robert, SAWG tests are not relevant until further JESD204 testing and debugging.

jbqubit, 2018-06-07:
I don't see any symptoms on my board suggesting JESD204b problems. I do see reproducible nonsense out of SAWG. Please take a look @jordens on Sayma in HK. Some additional examples...

```
from artiq.experiment import *

class SAWGTestSines(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()
        delay(1*ms)

        for sawg in self.sawgs:
            sawg.reset()
            delay(1*us)
            sawg.amplitude1.set(.5)
            sawg.frequency0.set(50*MHz)
            delay(100*ns)

        while True:
            delay(0.5*ms)
            self.ttl_sma_out.pulse(0.6*ms)
```

## frequency0.set(50*MHz)

https://www.youtube.com/watch?v=au4JqPSl5d0
This exact same pattern repeats from run to run and from boot to boot. This isn't a JESD issue. 

## frequency0.set(244*MHz)

https://www.youtube.com/watch?v=0TQgnauXb4M

## frequency0.set(280*MHz)
[image]
This exact same pattern repeats from run to run and from boot to boot. This isn't a JESD issue.

closed 2018-06-08 by jordens

jbqubit, 2018-06-08:
@jordens Still broken using 5b73dd86.

commit 6a7983cf (2018-06-08): conda: bump misoc

closes #1039
closes #1040
closes #1022
closes #1058
closes #1044