m-labs/artiq #1048: SUServo channel number confusion

cjbe, 2018-06-04:
The suservo DDS channel numbers seem to be in a nonstandard order: suservo channel 7 is urukul0 channel 0, whereas I would experiment channel 7 to be urukul1 channel 3.

I am using the suservo example experiment (https://github.com/m-labs/artiq/blob/master/artiq/examples/kasli_suservo/repository/suservo.py) and the following phy incantation:
```python
eem.SUServo.add_std(
    self, eems_sampler=(5, 4),
    eems_urukul0=(1, 0), eems_urukul1=(3, 2),
    t_rtt=15+4)
```

jordens, 2018-06-05:
Are you sure? Do you mean both profile settings and rf switch of `suservo0_ch0` end up on `urukul1, ch3`? I just checked again with the code in master (gateware and example) and it's fine there.

hartytp, 2018-06-05:
FWIW, I didn't see this in my tests, although they do predate the new eem.SUServo code.

cjbe, 2018-06-05:
@jordens this was a experiment / device_db mixup at my end - not an Artiq problem.

closed 2018-06-05 by cjbe