m-labs/artiq #1050: Coreanalyser exception with Kasli master-satellite

cjbe, 2018-06-05:
Using a Kasli master and satellite, I see the following exception most of the time artiq_coreanalyser is run while a kernel is running:
```
$ artiq_coreanalyzer -w dump
WARNING:artiq.coredevice.comm_analyzer:unable to determine DDS sysclk
Traceback (most recent call last):
  File "/home/ion/miniconda3/envs/artiq/bin/artiq_coreanalyzer", line 11, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_coreanalyzer')()
  File "/home/ion/scratch/artiq/artiq/frontend/artiq_coreanalyzer.py", line 57, in main
    decoded_dump)
  File "/home/ion/scratch/artiq/artiq/coredevice/comm_analyzer.py", line 543, in decoded_dump_to_vcd
    channel_handlers[message.channel].process_message(message)
  File "/home/ion/scratch/artiq/artiq/coredevice/comm_analyzer.py", line 384, in process_message
    while self._reads[0].rtio_counter < message.timestamp:
IndexError: list index out of range
```

sbourdeauducq, 2018-06-05:
That's with SPI, I suppose?

cjbe, 2018-06-05:
yes - there are SPI cores on both master and satellite (https://github.com/cjbe/artiq/blob/master/artiq/gateware/targets/kasli_hoa2.py)

sbourdeauducq, 2018-06-05:
And this is a problem with DRTIO and not just with SPI?

cjbe, 2018-06-06:
I have no evidence that this is to do with DRTIO, but I have not managed to quickly reproduce this problem on a setup without DRTIO.

jordens, 2018-06-06:
@cjbe This should fix it. Could you try again?

closed 2018-06-06 by jordens

cjbe, 2018-06-06:
@jordens that seems to fix it

commit 38971d13 (2018-06-06): comm_analyzer: fix data without any spi reads

closes #1050