m-labs/artiq #1055: Sayma: check HMC7043 output phases status on init

hartytp, 2018-06-05:
We should probably read back the "outputs phases status" bit from the HMC7043 before we return from init (see the typical programming sequence section, point 10). This verifies that the phases are stable before we unmute the outputs (i.e. no glitches).

I interpret that as meaning that we should read back register 0x7D, potentially after setting an appropriate mask in register 0x71, although maybe I've misunderstood that. Anyway, currently 0x7d  always seems to be 0x00, indicating that the phases aren't stable or that I'm doing something wrong.

As a low priority, would you mind looking into this @sbourdeauducq ir @enjoy-digital

hartytp, 2018-07-02:
Don't know.

Have you tried reading the phases statuses bit on your board?

sbourdeauducq, 2018-07-04:
Also 0x00.

hartytp, 2018-07-04:
Thanks for checking that.

So, either we're not using this bit correctly or the phases aren't settling properly.

enjoy-digital, 2018-07-05:
What's are you writing to register 0x71? Are you setting bit 4, 2 & 1?

sbourdeauducq, 2018-07-05:
Well, you have the code... we're not touching 0x71.

davidcarne, 2018-07-07:
So, I've been taking a bit of a look at this. @sbourdeauducq successfully nerd-sniped me on it.

The exact reseeding behaviour is complex, poorly documented, and in some cases, contradictory. The current codebase leaves the "RF reseeder" (reg 3, bit 5) enabled as is the default, which means the [SYSREF FSM is stuck in reset](https://ez.analog.com/thread/93998-hmc7043-sysref-fsm-state-question) ergo the sync can't complete (reg 0x91 stays at the RESET state).

However, even if we fix this, 0x7D still doesn't indicate phase stable. [Apparently 0x7D is broken](https://ez.analog.com/thread/97865-hmc7043s-rfsync-cant-work-properly) so we might need to change to checking stability via the GPO. I'll test a bit further down these avenues and determine whether the GPO or the 0x91 FSM status register is showing the SYSREF channels as synchronized.

Various analog devices forum posts suggest that we may want to ensure the RFSYNC inputs are forced to logic-zero if we aren't using them. This might require a HW change for a latter board rev.

sbourdeauducq, 2018-07-07:
> we may want to ensure the RFSYNC inputs are forced to logic-zero if we aren't using them

In accordance with the datasheet, those pins are AC-coupled on the board, so it is unclear to me how to connect them to a constant 0. Do those forum posts have the answer?

hartytp, 2018-07-08:
> In accordance with the datasheet, those pins are AC-coupled on the board, so it is unclear to me how to connect them to a constant 0. Do those forum posts have the answer?

Where do you see that? AFAICT, it's fine to tie them low by pulling P to GND and N to 3V3. Just make sure to set the biasing and termination appropriately (i.e. no biasing, no termination). See the clock input network section of the data sheet (which says that inputs are "typically AC coupled" not "must be AC coupled).

hartytp, 2018-07-08:
One thing to be careful about is that sometimes these input stages have diodes to limit the maximum differential input signal. So, it's worth making sure that any DC pull ups/downs are via 10k resistors to limit any currents (you can measure the voltage drop to check if any diodes are conducting), If you do that, you might as well enable the common-mode voltage setting resistors to ensure that you're biasing around the right quiescent point.

sbourdeauducq, 2018-07-09:
[image]

sbourdeauducq, 2018-07-09:
So we can work the other way: use documented termination resistor settings (e.g. 100 ohm), then one resistor to 3.3V one one pin, and one to GND on the other, with values that produce the correct voltages at the two ends of the termination resistor. Bonus, we can simply use a DMM to check that the termination settings of this horrible HMC7043 are correct. Considering how obscure this is again, it might be worth reviewing the CLKIN termination.

hartytp, 2018-07-09:
> I don't even see why they suggest RFSYNCIN should be AC-coupled, in some cases you may just send one pulse there, not an AC signal...

My understanding is that the anticipated "normal" use for this IC is to have a series of pulses on the RF sync input (indeed, it even says that this is only the "normal" usage in the text I quoted). Then AC coupling works best as it ensures that the signal swing is around the optimal CM voltage to give low jitter (remember it has to have low enough jitter to work up to be sampled of a several GHz clock). For single pulses, or just pulling it low, obviously AC coupling doesn't work.

In any case, the documentation seems poor and incorrect, so let's just try it and see. So long as the pull ups limit the current to something sensible we won't break anything.

>  Note "The input termination network is configurable to 100 Ω, 200 Ω, and 2kΩ differentially", but cannot be completely disabled AFAICT. It's also unclear how to configure this.

AFAICT, it's done with `0x000B` but the mapping between those bits and the settings they control is a bit opaque. I'm not sure I fully understand how this works, but:
- `Enable ac coupling input mode` I believe that this enables the common-mode voltage setting resistors
- `Enable LVPECL input mode` - I believe that this enables the 50 internal termination resistor on figure 48. It's not totally clear to me, but this may only be intended for DC-coupled LVPECL inputs. 
- `By closing the 50 Ω termination switch (see Figure 48), the network also serves as the termination system for an LVPECL driver.`
- `High-Z input enable` I assume that this enables the 2k differential termination (does that mean that the 200R termination is enabled if this register is set to 0x0?)

> So we can work the other way: use documented termination resistor settings (e.g. 100 ohm), then one resistor to 3.3V one one pin, and one to GND on the other, with values that produce the correct voltages at the two ends of the termination resistor.

Just be careful not to exceed the internal 2V8 supply that these pins are powered from.

Also, reading the docs (particularly the HMC7044, which is more detailed) it looks like these buffers are somewhat fussy over common-mode input voltages. So, I think it would be best to use the "AC-coupling" mode to enable the on-chip biassing and then use some weak-pull ups/downs to set the required logic levels (we only need to change the input voltage by a few hundred mV).

> Considering how obscure this is again, it might be worth reviewing the CLKIN termination.

Yes, I'm beginning to think our settings here are wrong. I don't think we should be using LVPECL mode.

> r. Bonus, we can simply use a DMM to check that the termination settings of this horrible HMC7043 are correct

That's probably a good thing to do. But, I'd feel much more comfortable if we could speak to someone who actually understands it instead of trying to reverse engineer it. Maybe post something on the ADI forum?

hartytp, 2018-07-09:
@sbourdeauducq also, note that [here](https://ez.analog.com/thread/93030-the-problems-of-hmc7043) an ADI guy says that the RFSYNCIN **should** be dc-coupled...

hartytp, 2018-07-09:
Input buffer modes: https://ez.analog.com/message/296858-hmc7043-input-buffer-mode

hartytp, 2018-07-09:
> The exact reseeding behaviour is complex, poorly documented, and in some cases, contradictory. The current codebase leaves the "RF reseeder" (reg 3, bit 5) enabled as is the default, which means the SYSREF FSM is stuck in reset ergo the sync can't complete (reg 0x91 stays at the RESET state).

@sbourdeauducq we should also definitely check the SYSREF FSM state during init. Looking at the reseeder request description on the data sheet: 

>Requests the centralized resync timer and FSM to reseed any of the output dividers that are programmed to pay attention to sync events. This signal is rising edge sensitive, and is only acknowledged if the resync FSM has completed all events (has finished any previous pulse generator and/or sync events, and is in the done state (SYSREF FSM State[3:0] = 0010)"

hartytp, 2018-07-09:
I looks like we should also explicitly disable the SYSREF timer once we've finished synchronisation. https://ez.analog.com/message/302525-hmc7043-subharmonics

sbourdeauducq, 2018-07-09:
We're not using the analog delay (which is noisy anyway) on the high-frequency clocks such as the DAC sample clocks. It is only used on the lower-frequency SYSREF channels.
We're not using any of this SYSREF/FSM/RFSYNCIN logic so the best way to deal with it is to disable it as much as possible.

jbqubit, 2018-07-09:
It might be worth opening an Analog Devices service request on this. They may be willing to do a design review of HMC7043 as used on Sayma.

commit 48438323 (2018-07-11): hmc7043: check phase status on init. Closes #1055

Troubleshooting by David.
closed 2018-07-11 by sbourdeauducq