m-labs/artiq #1057: kasli variants don't meet timing

jordens, 2018-06-06:
Several kasli variants don't meet timing anymore in the majority of the compilation attempts (HUB, LUH, SUServo last I checked).
c.f. #891

sbourdeauducq, 2018-07-09:
All three now seem to pass timing with Vivado 2018.2 and b0a47ddc + 83c83547 on MiSoC.

hartytp, 2018-09-25:
@sbourdeauducq I tried building a DRTIO master (master, not switching) with a 16-channel SUServo and found some quite large timing violations. 

```python
class ServoMaster(_MasterBase):
    def __init__(self, hw_rev=None, **kwargs):
        if hw_rev is None:
            hw_rev = "v1.1"
        _MasterBase.__init__(self,
                             hw_rev=hw_rev,
                             rtio_clk_freq=125e6,
                             **kwargs)

        # Use the higher speed grade FPGA, -3, instead of the default -2
        self.platform.device = "xc7a100t-fgg484-3"

        if hw_rev == "v1.0":
            # EEM clock fan-out from Si5324, not MMCX
            self.comb += self.platform.request("clk_sel").eq(1)

        self.rtio_channels = []

        # EEM1/0: Sampler, EEM3/2: Urukul, EEM5/4: Urukul
        eem.SUServo.add_std(
            self, eems_sampler=(1, 0),
            eems_urukul0=(3, 2), eems_urukul1=(5, 4))

        # EEM 6:DIO
        eem.DIO.add_std(self, 6,
                        ttl_serdes_7series.Output_8X,
                        ttl_simple.Output)

        self.config["HAS_RTIO_LOG"] = None
        self.config["RTIO_LOG_CHANNEL"] = len(self.rtio_channels)
        self.rtio_channels.append(rtio.LogChannel())

        self.add_rtio(self.rtio_channels)

        pads = self.platform.lookup_request("sampler1_adc_data_p")
        self.platform.add_false_path_constraints(
            pads.clkout, self.drtio_transceiver.gtps[0].txoutclk)
        self.platform.add_false_path_constraints(
            pads.clkout, self.crg.cd_sys.clk)
```

Without the higher-speed FPGA, this gives a lot of ~1ns timing violations. With the faster FPGA, it builds fine.

Side note, do the false path constrains look correct, or am I missing something?

sbourdeauducq, 2019-06-11:
Now that we have a lot more variants, this problem is coming back :(

sbourdeauducq, 2019-09-09:
Haven't seen this again recently...

commit ea71a049 (2018-08-27): conda: bump misoc. Closes #1057
closed 2019-09-09 by sbourdeauducq