m-labs/artiq #1058: Sayma SAWG frequency0.set(301*MHz) produces variable amplitude signal

marmeladapk, 2018-06-06:
[Video showing the issue](https://wutwaw-my.sharepoint.com/:v:/g/personal/252842_pw_edu_pl/EVjxDPCBVhlHj7ysazwp0FoBN7jETTSDm9pTmB3mMyd2NQ?e=BdTUsJ)

Artiq: c28fe47, Misoc: 86c6fcb, Migen: 6425844, jesd204b: 25fd79d

Experiment:
```
from artiq.experiment import *

class SAWGTest(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_sma_out")
        self.sawgs = [self.get_device("sawg"+str(i)) for i in range(8)]

    @kernel
    def run(self):
        self.core.reset()

        for sawg in self.sawgs:
            delay(1*ms)
            sawg.amplitude1.set(.8)
            # Do not use a sub-multiple of oscilloscope sample rates.
            sawg.frequency0.set(301*MHz)

        while True:
            delay(0.5*ms)
            self.ttl_sma_out.pulse(0.5*ms)
```

UART log:
```
Booting from flash...
Starting firmware.
[     0.000007s]  INFO(runtime): ARTIQ runtime starting...
[     0.003882s]  INFO(runtime): software version 4.0.dev+1116.gc28fe471
[     0.010233s]  INFO(runtime): gateware version 4.0.dev+1110.g07d4145a
[     0.016601s]  INFO(runtime): log level set to INFO by default
[     0.022308s]  INFO(runtime): UART log level set to INFO by default
[     0.028456s]  INFO(board_artiq::serwb): waiting for AMC/RTM serwb bridge to be ready...
[     0.062135s]  INFO(board_artiq::serwb): done.
[     0.065166s]  INFO(board_artiq::serwb): RTM to AMC Link test
[     1.547208s]  INFO(board_artiq::serwb): 0 errors
[     1.550511s]  INFO(board_artiq::serwb): AMC to RTM Link test
[     3.032560s]  INFO(board_artiq::serwb): 0 errors
[     3.035859s]  INFO(board_artiq::serwb): Wishbone test...
[     4.967727s]  INFO(board_artiq::serwb): 0 errors
[     4.971034s]  INFO(board_artiq::serwb): AMC serwb settings:
[     4.976583s]  INFO(board_artiq::serwb):   bitslip: 27
[     4.981619s]  INFO(board_artiq::serwb):   ready: 1
[     4.986395s]  INFO(board_artiq::serwb):   error: 0
[     4.991170s]  INFO(board_artiq::serwb): RTM serwb settings:
[     4.996736s]  INFO(board_artiq::serwb):   bitslip: 16
[     5.001772s]  INFO(board_artiq::serwb):   ready: 1
[     5.006547s]  INFO(board_artiq::serwb):   error: 0
[     5.011557s]  INFO(board_artiq::serwb): RTM gateware version 4.0.dev+1116.gc28fe471
[     5.018963s]  INFO(runtime): press 'e' to erase startup and idle kernels...
[     6.018006s]  INFO(runtime): continuing boot
[     6.021065s]  INFO(board_artiq::hmc830_7043::hmc830): HMC830 found
[     6.027133s]  INFO(board_artiq::hmc830_7043::hmc830): loading configuration...
[     6.034641s]  INFO(board_artiq::hmc830_7043::hmc830):   ...done
[     6.040230s]  INFO(board_artiq::hmc830_7043::hmc830): waiting for lock...
[     6.047035s]  INFO(board_artiq::hmc830_7043::hmc830):   ...locked
[     6.053080s]  INFO(board_artiq::hmc830_7043::hmc7043): enabling hmc7043
[     6.059955s]  INFO(board_artiq::hmc830_7043::hmc7043): HMC7043 found
[     6.066033s]  INFO(board_artiq::hmc830_7043::hmc7043): loading configuration...
[     6.075314s]  INFO(board_artiq::hmc830_7043::hmc7043):   ...done
[     6.080368s]  INFO(board_artiq::ad9154): AD9154-0 found
[     6.095507s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     6.108435s]  INFO(board_artiq::ad9154):   ...done
[     6.182324s]  INFO(board_artiq::ad9154): AD9154-0 running PRBS test...
[     7.188701s]  INFO(board_artiq::ad9154):   ...passed
[     7.192361s]  INFO(board_artiq::ad9154): AD9154-0 running STPL test...
[     7.199150s]  INFO(board_artiq::ad9154):   c0 errors: 0
[     7.204355s]  INFO(board_artiq::ad9154):   c1 errors: 0
[     7.209565s]  INFO(board_artiq::ad9154):   c2 errors: 0
[     7.214774s]  INFO(board_artiq::ad9154):   c3 errors: 0
[     7.219698s]  INFO(board_artiq::ad9154):   ...passed
[     7.224647s]  INFO(board_artiq::ad9154): AD9154-0 SYSREF scan:
[     7.504737s]  INFO(board_artiq::ad9154):   phase: 26, sync error: 481
[     8.159976s]  INFO(board_artiq::ad9154):   phase: 90, sync error: 482
[     8.165115s]  INFO(board_artiq::ad9154):   phase min: Some(26), phase max: Some(89)
[     8.172757s]  INFO(board_artiq::ad9154): AD9154-0 setting SYSREF phase to 88
[     8.190141s]  INFO(board_artiq::ad9154): AD9154-0 initializing...
[     8.203033s]  INFO(board_artiq::ad9154):   ...done
[     8.277242s]  INFO(board_artiq::ad9154): AD9154-1 found                           
[     8.291440s]  INFO(board_artiq::ad9154): AD9154-1 initializing...                 
[     8.304292s]  INFO(board_artiq::ad9154):   ...done                                
[     8.381706s]  WARN(board_artiq::ad9154): AD9154-1 config attempt #0 failed (bad SYNC), retrying
[     8.399409s]  INFO(board_artiq::ad9154): AD9154-1 initializing...                 
[     8.412264s]  INFO(board_artiq::ad9154):   ...done
[     8.486140s]  INFO(board_artiq::ad9154): AD9154-1 running PRBS test...
[     9.492508s]  INFO(board_artiq::ad9154):   ...passed
[     9.496166s]  INFO(board_artiq::ad9154): AD9154-1 running STPL test...
[     9.502951s]  INFO(board_artiq::ad9154):   c0 errors: 0
[     9.508161s]  INFO(board_artiq::ad9154):   c1 errors: 0
[     9.513371s]  INFO(board_artiq::ad9154):   c2 errors: 0
[     9.518581s]  INFO(board_artiq::ad9154):   c3 errors: 0
[     9.523503s]  INFO(board_artiq::ad9154):   ...passed
[     9.528451s]  INFO(board_artiq::ad9154): AD9154-1 SYSREF scan:
[     9.788214s]  INFO(board_artiq::ad9154):   phase: 24, sync error: 481
[    10.443470s]  INFO(board_artiq::ad9154):   phase: 88, sync error: 482
[    10.448607s]  INFO(board_artiq::ad9154):   phase min: Some(24), phase max: Some(87)
[    10.456248s]  INFO(board_artiq::ad9154): AD9154-1 setting SYSREF phase to 88
[    10.473635s]  INFO(board_artiq::ad9154): AD9154-1 initializing...
[    10.486490s]  INFO(board_artiq::ad9154):   ...done
[    10.560358s]  INFO(board_artiq::hmc542): card 0 channel 0 set to 4 dB
[    10.567602s]  INFO(board_artiq::hmc542): card 0 channel 1 set to 4 dB
[    10.574838s]  INFO(board_artiq::hmc542): card 1 channel 0 set to 4 dB
[    10.582074s]  INFO(board_artiq::hmc542): card 1 channel 1 set to 4 dB
[    10.589309s]  INFO(board_artiq::hmc542): card 2 channel 0 set to 4 dB
[    10.596544s]  INFO(board_artiq::hmc542): card 2 channel 1 set to 4 dB
[    10.603780s]  INFO(board_artiq::hmc542): card 3 channel 0 set to 4 dB
[    10.611015s]  INFO(board_artiq::hmc542): card 3 channel 1 set to 4 dB
[    10.618303s]  INFO(runtime): using MAC address 34-45-32-12-ff-22
[    10.623090s]  INFO(runtime): using IP address 192.168.95.176
[    10.630310s]  INFO(runtime::mgmt): management interface active
[    10.643619s]  INFO(runtime::session): accepting network sessions
[    10.657822s]  INFO(runtime::session): running startup kernel
[    10.662287s]  INFO(runtime::session): no startup kernel found
[    10.668028s]  INFO(runtime::session): no connection, starting idle kernel
[    10.674893s]  INFO(runtime::session): no idle kernel found
[    20.221818s]  INFO(runtime::session): new connection from 192.168.95.122:48926
[    20.259751s]  INFO(runtime::kern_hwreq): resetting RTIO
```

jbqubit, 2018-06-06:
Link to video doesn't work.

marmeladapk, 2018-06-06:
@jbqubit I have verified that it works on another computer and on mobile phone which wasn't connected to the same network.

Perhaps [youtube](https://youtu.be/Zy5CWWd2EXM) will work for you?

marmeladapk, 2018-06-06:
@jordens For every frequency above 300 MHz the effect is the same. Well if Sayma doesn't support frequencies above 300 MHz then there's nothing to talk about.

jordens, 2018-06-06:
@marmeladapk The sample rate is 600 MHz. It supports frequencies above 300 MHz just fine. In the sense of Nyquist images.
Could you try some of the experiments from #1040, #1039?

jbqubit, 2018-06-06:
> It supports frequencies above 300 MHz just fine. In the sense of Nyquist images.

Ya, OK. But we're still struggling to understand what's going on in the 1st Nyquist zone.

commit 6a7983cf (2018-06-08): conda: bump misoc

closes #1039
closes #1040
closes #1022
closes #1058
closes #1044
closed 2018-06-08 by jordens