m-labs/artiq #1060: Variant should be added to the identifier (gateware, firmware, RTM)

jordens, 2018-06-07:
c.f. #1057

sbourdeauducq, 2018-06-29:
The RTM gateware is the same for all variants.

sbourdeauducq, 2018-06-29:
That's about the ARTIQ variant, not hardware versions.

gkasprow, 2018-06-29:
OK, but to get rid of confusion when next HW releases arrives, we should check the board ID before loading ARTIQ.

sbourdeauducq, 2018-06-29:
Alternatively: avoid breaking compatibility, or deprecate the current set of boards - they have plenty of bugs and fragile reworks anyway.

hartytp, 2018-06-29:
Let's depreciate them. Maintaining software support for old boards gets very expensive and isn't worth it for a handful of quite hacked prototypes.

gkasprow, 2018-06-29:
I mean protection feature that won't enable loading i.e. old firmware to new boards. Then user will get clear message instead of frustration that something does not work.

commit b27fa896 (2018-07-15): add variant in identifier string

Also add without-sawg suffixes on Sayma.

Closes #1060
Closes #1059
closed 2018-07-15 by sbourdeauducq